solvespace/solvespace #887: libdxfrw Hatch-Spline Parser does not conform to DWG Spec

Akaame, 2021-01-10:
### System information

- SolveSpace version: None
- Ubuntu 20.04, GCC 10.2,0

### Actual behavior

As per the issue in LibreCAD's libdxfrw, available [here](https://github.com/LibreCAD/libdxfrw/issues/18), there seems to be several errors in Hatch-Spline parser code that parses spline control point count wrong which in turn cause large allocations which hogs large amounts of memory (larger than 40GBs). 

### Expected behavior

Spline control points must be parsed correctly.

### Lines in Question

https://github.com/solvespace/libdxfrw/blob/0b7b7b709d9299565db603f878214656ef5e9ddf/drw_entities.cpp#L1961

https://github.com/solvespace/libdxfrw/blob/0b7b7b709d9299565db603f878214656ef5e9ddf/drw_entities.cpp#L1964

[Link to original repo.](https://sourceforge.net/projects/libdxfrw/)

[Link to DWG specification from Open Design Alliance](https://www.opendesign.com/files/guestdownloads/OpenDesign_Specification_for_.dwg_files.pdf)

Section 20.4.75 describes the HATCH structure in DWG file. In Page 196 you can see that parsing of ncontrol comes before parsing of individual knots. Following that section individual control points are denoted by 2RD(2 Raw Double) instead of 3BD

I would appreciate it if someone could validate my findings. 
Regards.

phkahler, 2021-01-11:
Can this be reproduced in SolveSpace? Looks like a patch is available if it's a problem for us.

Akaame, 2021-01-11:
I will try and reproduce this behaviour and see if the patch in question solves the problem.

Akaame, 2021-01-16:
Hello there,

Sorry I could not get back to this thread. The "day job" was kind of hectic this week. 

I was able to replicate the issue on solvespace master and solvespace edge.

### Replication steps.

1. Download the DWG file in the original issue.
2. Open your favorite system monitor software
3. File -> Import -> Import downloaded file from step 1
4. Watch your memory consumed to its last bit :smile: 

### Additional

I applied the patch and import was successful. However, I am not sure if the import result is legit or not(since I have no experience with CAD software).

Cheers.