solvespace/solvespace #888: Feature Request: Trackball Rotation

giannissc, 2021-01-12:
I was wondering if it would be useful to have trackball rotation (as described [here](https://www.mattkeeter.com/projects/rotation/).

It seems that some people have trouble with the existing rotation system #325 so maybe that would solve it or at least offer another option for people who don't like the existing system.

Thanks for all your good work!

ruevs, 2021-01-12:
There is turntable navigation (find the option on the configuration screen).
There is also `CTRL+middle button drag` - try it it's useful.

phkahler, 2021-01-14:
So the turntable is kind of nice. That CTRL-middle-mouse is terrible. What is that even doing? It seems to affect the turntable somehow - messing it all up! Does anyone use that?

ruevs, 2021-01-14:
`CTRL+middle button drag` rotates around the center of the view port in the plane of the screen. It is very useful in the "normal" mode - I use it all the time.

When turntable navigation on it behaves stupidly - I do not know if anyone ever considered this - turntable navigation was merged in fairly recently (e67f967933e Author: Dynamo Dan 2016-12-29 16:34:30 Committer: whitequark 2019-05-23 18:45:15)

vespakoen, 2021-02-02:
It would be nice to have Turntable nav on all axes (I experimented with this once but failed to get the right transform for one of the axes)

Ah, just found it: https://github.com/solvespace/solvespace/compare/master...vespakoen:turntable-axes

I was also thinking that locking the rotation to a single axis while rotation an assembly could be helpful.

I am not sure what is the best way yet but this area could really use some improvement.