solvespace/solvespace #890: Call for Translators

phkahler, 2021-01-16:
We are very close to releasing SolveSpace version 3.0 (final) but we'd like the language translations updated. I'm sure it's no fun to have Chinese menus with a stray English item thrown in the middle for example.

There are several new "strings" in SolveSpace that currently have no translation to other languages. We need speakers of those languages to update the .po file for each language (using poedit for example). Languages in need of update are:

  * [ ] de-DE,0407,Deutsch
  * [ ] en-US,0409,English (US)
  * [ ] fr-FR,040C,Français
  * [ ] ru-RU,0419,Русский
  * [ ] uk-UA,0422,Українська
  * [ ] zh-CN,0804,简体中文

Other languages can be added as well, but we'd like translations as near-complete as possible, which can take a lot of time.
The master .pot file is up to date in:  /res  folder
Languages  .po files can be found in the /res/locales folder. Pull requests welcome.

Code should be taken from the master branch, the latest "edge" release, or the (very soon) RC2 release.

phkahler, 2021-01-16:
While anyone can do this, I'd like to ping the following people who have done things for us before.

@lomatus  if you're still around, could you update CN?
@rurban  any chance you're available?
@Symbian9 for UA?
@Evil-Spirit for RU?

Symbian9, 2021-01-16:
I'm In for update `uk_UA`.

rurban, 2021-01-16:
Yes

rurban, 2021-01-17:
I am ready with de_DE, but got a question:
What does the new dimSolidModel menu option mean?
"Dim Solid for Sketch Groups"
I dont see it from the code. 
I only see groupmesh.cpp: DrawMesh fillFront.color = DIM_SOLID. So it's a default color.

The src and translation is a bit rough there. 
Does it mean diminuated or related to a Dimension? Or something else?

The git commit says: "Makes shadowing the solid optional for sketch in 2d/3d groups. Handy for making dimensioned drawings by putting dims in their own group. #834"
The menu item is not really understandable to me yet.

ruevs, 2021-01-17:
The oprion is on the view menu. By default it is on. This is the original behaviour.

If it is switched off then solids from previous groups bill not be "dimmed" (as ln lightness, darkened). Instead they will remain the color that is assigned to them and displayed when their group is active.

ruevs, 2021-01-17:
The style that is forced on "dimmed" models is s00d-#def-dim-solid
[image]

ruevs, 2021-01-17:
"Fade Previous Solid Groups"
Is better in my opinion as well.

rurban, 2021-01-17:
Ok, so it's not related to dimensions, rather if to fade (shadow) solids or not.
So I changed it to "Feste Farbe für Festkörper in Skizzen Gruppen". German is a horrible language, really

phkahler, 2021-01-17:
Wow! I thought that may need clarification. It seemed hard to describe in a short menu text.
Dim
Darken
Shadow
Fade

I think I might prefer Darken for english.

ruevs, 2021-01-17:
OK I'll make a pull request with Darken.

ruevs, 2021-01-17:
@phkahler "Darken Inactive Solid Groups"
or "Darken Previous Solid Groups" ?

ruevs, 2021-01-17:
"Darken Solids in Inactive Groups" ?
(taking a hint from the German)

ruevs, 2021-01-17:
Finally "Darken Inactive Solids" to keep it shorter than all others.

phkahler, 2021-01-17:
> Finally "Darken Inactive Solids" to keep it shorter than all others.

I think that works. Remember it only happens for sketch-in-2d or sketch-in-3d groups, not extrudes, lathe, etc...

Big questions, should it be changed in the code? That means all translations will need adjustment but it should help translators understand what it means.

ruevs, 2021-01-17:
I'll change it in the code and update the translations.

ruevs, 2021-01-17:
> 
> 
> > Finally "Darken Inactive Solids" to keep it shorter than all others.
> 
> I think that works. Remember it only happens for sketch-in-2d or sketch-in-3d groups, not extrudes, lathe, etc...

I know... but it's too verbose... and still confusing.

phkahler, 2021-01-17:
What about:
Darken solids in sketch groups

ruevs, 2021-01-17:
> 
> 
> What about:
> Darken solids in sketch groups

It's confusing, because someone will think "There can be solids in sketch groups?! How do I do that?"

phkahler, 2021-01-17:
Darken solids when sketching?

phkahler, 2021-01-17:
Never mind, I merged the PR. So the text translators will see is better, and the english version is also better than it was.

ruevs, 2021-01-17:
> 
> 
> Darken solids when sketching?
> Darken solids while sketching?

"Darken solids when sketching" just "fits" I can change it if you want.

```C++
{ 1, N_("Show Snap &Grid"),             Command::SHOW_GRID,        '>',     KC, mView  },
{ 1, N_("Darken solids when sketching"),Command::DIM_SOLID_MODEL,  0,       KC, mView  },
{ 1, N_("Use &Perspective Projection"), Command::PERSPECTIVE_PROJ, '`',     KC, mView  },
```

phkahler, 2021-01-17:
@ruevs No, I'm good with it as is. Let's not create churn while others are translating ;-)

I do like your wording though.

Symbian9, 2021-01-18:
> "Darken Solids When Sketching" just "fits" I can change it if you want.
> 
> ```
> { 1, N_("Show Snap &Grid"),             Command::SHOW_GRID,        '>',     KC, mView  },
> { 1, N_("Darken Solids When Sketching"),Command::DIM_SOLID_MODEL,  0,       KC, mView  },
> { 1, N_("Use &Perspective Projection"), Command::PERSPECTIVE_PROJ, '`',     KC, mView  },
> ```

@ruevs @phkahler , I'm voting for `Fade Solids When Sketching`, which fits more than `Darken Solids When Sketching`.

As it possible to set , "Darken" is true only for *default* setting, but it could be customized to became "Lighter" (see screenshot), so "Darken" really is NOT true here.

That is why I also requested renaming of `s00d-#def-dim-solid` to  `s00d-#def-fade-solid`

[image]

ruevs, 2021-01-18:
The rename of the style is not size (backward compatibility) but you are right that it is really:
"Assign the  s00d-#def-dim-solid Style to Solids When Sketch Groups are Active"

In my opinion "fade" also does not convey the exact meaning.

Symbian9, 2021-01-18:
> Hopefully a user who has gotten to changing this style will know what he is doing and will not be confused.

Guess, we should add deeply explanation notes on styles into SolveSpace Tutorial & Refernce docs.

ruevs, 2021-01-18:
> 
> 
> > Hopefully a user who has gotten to changing this style will know what he is doing and will not be confused.
> 
> Guess, we should add deep explanation notes on styles into SolveSpace Tutorial & Refernce docs.

The re is a [section on "Line Styles" in the reference](https://solvespace.com/ref.pl) and styles usage is described in a few more relevant places. But a section on applying styles to solids certainly would be useful.

Evil-Spirit, 2021-01-22:
[ru_RU.zip](https://github.com/solvespace/solvespace/files/5855565/ru_RU.zip)

kk6mrp, 2021-03-05:
The last three translations still needed were originally from: @patrovite, @Symbian9, and @youzhou2020

Symbian9, 2021-03-06:
`uk_UA` actually in work!

<details>
    <summary>WIP (hopefully, it would be complete in two days, til *March 8th*)<summary>
[image]
<details>

mhalil, 2021-03-09:
Hi,
I translated the en_EN.po file into Turkish
[tr_TR.po.zip](https://github.com/solvespace/solvespace/files/6111158/tr_TR.po.zip)
[image]

phkahler, 2021-03-17:
@mhalil  Thank you. I hope to merge this very soon.

closed 2021-04-04 by phkahler