solvespace/solvespace #892: Update toolbar icons

phkahler, 2021-01-16:
Looking for some of the new icon bitmaps from the discussion in #857 since we don't have vector icons yet. I know @Symbian9  and @jkrei0 have some newer version. Can we somehow pick the best of what's been created so far and get them in?

In particular I'd like the isometric-view icon to actually be isometric and I know there were some in the thread ;-)  We also merged a new Revolve and Helix icon from that discussion, so this request is purely to update the image files with no code changes.

phkahler, 2021-01-17:
So @ruevs made a pull request #897 with more icons pulled from the previous discussion. I made some comments in the linked PR. Question now is: Should I merge that PR as a new starting point? There are before and after images in the PR.

What's a good PNG editor on Linux? I want to fix those red X myself ;-)   240, 32, 32 is better with no transparent.

ruevs, 2021-01-17:
[image] Working on an "isometric" icon. But AN isometric view can look like the current icon. Just not our `F3` 120° one.
For example open the attached file and press `W` ;-) Then press `F3`...
[IsometricViewVsOurIcon.zip](https://github.com/solvespace/solvespace/files/5826848/IsometricViewVsOurIcon.zip)

I agree with most of what @phkahler [said in the pull request](https://github.com/solvespace/solvespace/pull/897#issuecomment-761880092) and will try to fix the offending ones.

I think you should not merge the pull request. Let us make the icons "mostly pleasing to most people" first :-D

A good PNG editor for pixel peeping work is IrfanView - run it under Wine if you must ;-)

ruevs, 2021-03-03:
I updated the icons again. Please see the pull request for details. #897

closed 2021-03-06 by phkahler