solvespace/solvespace #893: arm build issue, rc1 on Debian

rpavlik, 2021-01-16:
### System information

- **SolveSpace version:** 3.0rc1
- **Operating system:**  Debian testing

https://lists.debian.org/debian-science/2021/01/msg00034.html

vespakoen, 2021-01-16:
Does it work when updating mimalloc to the latest master?

Are there logs of those builds somewhere?

ruevs, 2021-01-17:
https://github.com/solvespace/solvespace/issues/772#issuecomment-761133282

phkahler, 2021-01-18:
@rpavlik  Is it a build issue or failing tests?  @ruevs build on a RaspberryPi recently, mentioned here: https://github.com/solvespace/solvespace/issues/876

rpavlik, 2021-01-18:
Log is here. Looks like only armel, not armhf (rpi) or arm 64. https://buildd.debian.org/status/fetch.php?pkg=solvespace&arch=armel&ver=3.0.rc1%2Brepack1-4&stamp=1610723096&raw=0

rpavlik, 2021-01-18:
```
 /usr/bin/cmake -E make_directory /<<PKGBUILDDIR>>/obj-arm-linux-gnueabi/res
/tmp/ccBLJB7o.s: Assembler messages:
/tmp/ccBLJB7o.s:10022: Error: selected processor does not support `yield' in ARM mode
cd /<<PKGBUILDDIR>>/obj-arm-linux-gnueabi/res && /usr/bin/cmake -E copy /<<PKGBUILDDIR>>/res/locales.txt /<<PKGBUILDDIR>>/obj-arm-linux-gnueabi/res/locales.txt
[ 23%] Copying resource locales/de_DE.po
cd /<<PKGBUILDDIR>>/obj-arm-linux-gnueabi/res && /usr/bin/cmake -E make_directory /<<PKGBUILDDIR>>/obj-arm-linux-gnueabi/res/locales
make[3]: *** [extlib/mimalloc/CMakeFiles/mimalloc-static.dir/build.make:150: extlib/mimalloc/CMakeFiles/mimalloc-static.dir/src/segment.c.o] Error 1
make[3]: Leaving directory '/<<PKGBUILDDIR>>/obj-arm-linux-gnueabi'
make[2]: *** [CMakeFiles/Makefile2:305: extlib/mimalloc/CMakeFiles/mimalloc-static.dir/all] Error 2
```

rpavlik, 2021-01-18:
Is Mimalloc required? Or is it just a "nice to have"?

phkahler, 2021-01-18:
> Is Mimalloc required? Or is it just a "nice to have"?

I'm not certain. I think it became necessary when we introduced OpenMP, so maybe try without that option and without Mimalloc for that arch.

This ARM talk go me thinking... I think we're OK but there is some code that assumes little-endian representation of floats (not doubles) in our .STL export function.

ruevs, 2021-01-18:
It did not become necessary, because of OpenMP.
It got introduced here

phkahler, 2021-01-18:
> Mimalloc did **_not_** become necessary, because of OpenMP.

Yes it did. I recall being quite impressed that whitequark went off on a tangent with memory allocators just to support some early OpenMP additions. The end result of all that was Mimalloc. IIRC we had either some leaks or sanitizer failures until that all got straightened out.

Or my memory might be failing me... That's possible.

vespakoen, 2021-01-18:
Looks like this issue is fixed on mimalloc's dev branch: https://github.com/microsoft/mimalloc/commit/33a10b48605f8bb419487a03125815ad6ee00a70

rpavlik, 2021-01-18:
Oh awesome we can just borrow that patch for Debian if nothing else.

rpavlik, 2021-01-19:
OK, I have borrowed that patch for a debian package patch, hopefully it works on armel (I don't have an armel build env set up offhand) https://salsa.debian.org/rpavlik/solvespace

ruevs, 2021-01-20:
Without knowing any details that patch looks strange to me on lines 285 and 290 the condition is the same `__ARM_ARCH__ >= 7`. That does not look right.

ruevs, 2021-01-20:
I commented on the commit https://github.com/microsoft/mimalloc/commit/33a10b48605f8bb419487a03125815ad6ee00a70#r46141051

ruevs, 2021-01-20:
Sorry my mistake. I did not parse the nesting looking at it on GitHub.
https://github.com/microsoft/mimalloc/commit/33a10b48605f8bb419487a03125815ad6ee00a70#r46141225

rpavlik, 2021-01-26:
OK, so this patch fixed the build on Debian. Not sure how we want to import it here: if we apply the patch locally, or wait for upstream to grab it, or...? https://salsa.debian.org/science-team/solvespace/-/blob/master/debian/patches/10_mimalloc_restrict_cpu_yield.patch

phkahler, 2021-01-26:
> Not sure how we want to import it here: if we apply the patch locally, or wait for upstream to grab it, or...?

Any ETA on upstream merging it? Then we can just bump our required mimalloc version.

We really are just giving some time hoping for translations now - 3 down 3 to go.

rpavlik, 2021-01-26:
No idea, upstream hasn't acted on it except for their CLA bot verifying it.

phkahler, 2021-02-01:
@rpavlik  mimalloc just pushed out a 2.0.0 version yesterday.

vespakoen, 2021-02-01:
Great! I will update my PR to use it and apply a temporary patch that (kindof) fixes a issue that I found with mimalloc on macOS arm64 (https://github.com/microsoft/mimalloc/issues/354)

ruevs, 2021-02-02:
Do you guys think that the update of mimalloc should be a separate pull request? Or at least a separate commit in the M1 pull request.

closed 2021-04-04 by phkahler