solvespace/solvespace #900: Text input fields override "Property Browser" window width size

Symbian9, 2021-01-18:
### System information

- **SolveSpace version:** `3.0~` https://github.com/solvespace/solvespace/commit/bd41485ebecce360baa729567273714caac1df6c
- **Operating system:** Debian 10 buster (MX Linux MX-19.3)

### Expected behavior

Text input fields should fit "Property Browser" window width & do NOT enlarge it.

FTR, 'In place' text input field behavior should be reworked & text input fields should alligned to the left border of "Property Browser" (additionally there should be an option to limit text input field width in number of characters as drop-down list as a range of: `10`, `20`, `50`, `100`)

### Actual behavior

Text input fields enlarges "Property Browser" window size

- Video: 

### Additional information

All this tested on 'clean install' with default configuration.

phkahler, 2021-01-18:
In addition to the input box resizing the text window, I believe the minimum width of the text window is too small. Given that there is a minimum (and I believe there should be) it should be wide enough to show all the toolbar icons. That change alone would prevent some of the unwanted resizing. It may also help if the color picker was positioned a little to the left so it doesn't require the window to be wider than the toolbar (minimum).

There will not be a configuration option for text input size, but we can do better than it is now.

I'm curious if this is Linux/GTK specific or if it happens on Window or macOS too. I suspect it does.

Symbian9, 2021-01-18:
> I'm curious if this is Linux/GTK specific or if it happens on Window or macOS too. I suspect it does.

JFTR, Beside input field should fit "Property Browser", guess input field should be also scrollable (for input/show text strings)

ruevs, 2021-01-18:
> In addition to the input box resizing the text window, I believe the minimum width of the text window is too small. Given that there is a minimum (and I believe there should be) it should be wide enough to show all the toolbar icons.

I disagree, I regularly resize the text window to narrower than the toolbar width. In fact I keep it this way most of the time.

Symbian9, 2021-01-18:
> I disagree, I regularly resize the text window to narrower than the toolbar width

Problem here is that minimum size of "Property Browser" window came from `2.0` where only 10 buttons was on toolbar (and it fits to minimal window size), but since `3.x` one additional button was added: *"Hide construction enteties"*, and now we has 11 buttons on toolbar.

So, this issue was came from `2.0` & it is UI **bug** —  in SolveSpace `3.0` minimal width of "Property Browser" should fit to 11 buttons (instead of 10 as it was in `2.0`)

ruevs, 2021-01-19:
I don't mind that it does not fit - I don't use those buttons so often. On the other hand the text window takes space on my screen all the time (I know about `Tab`) and it needs to be the smallest possible that will show the active group, DOF, and the beginning of the name.