solvespace/solvespace #901: Edit text string of "Text in TrueType Font" by double click in main window

Symbian9, 2021-01-18:
### System information

- **SolveSpace version:** `3.0~` https://github.com/solvespace/solvespace/commit/bd41485ebecce360baa729567273714caac1df6c
- **Operating system:** Debian 10 buster (MX Linux MX-19.3)

### Expected behavior

Open text input field if double click on "Text in TrueType Font" directly in main window (as it works for "Comment" constrain text input).

I suppose `text = {text_string} [edit]` should be removed. Leave only font choosing & other 

***Additionally***, reorganize  "TrueType Font Text" property tab layout by moving object property from bottom to top (ahead of fonts list) as next:

```
TRUETYPE FONT TEXT

FROM REQUEST   {request_id}
IN GROUP       {group_id}
IN WORKPLANE   {workplane_id}
IN STYLE       {style_id}

FONT = {font_name}

  select new font
     {list_of_available_fonts}
     ...
```

### Actual behavior

Actual behavior on editing text string of the "Text in TrueType Font" via "Property Browser" is ugly implementation.

### Additional information

phkahler, 2021-01-18:
I'm thinking "no" for the first part - double click to edit - because the text on the sketch can be of arbitrary size and orientation. Putting an edit control out there would not feel like directly editing and probably awkward.

If someone were really ambitious they might make a special edit-box capability out of the existing text entity. That could be painful when the sketch complexity is high and each character takes 2 seconds to appear. Making a text edit box can be extremely time consuming to do well - edit, insert, delete, cut, paste, focus issues...

Reordering the text window makes sense to me since the font list can be very long, effectively hiding the stuff at the end.

Symbian9, 2021-01-18:
> *If someone were really ambitious they might make a special edit-box capability out of the existing text entity. That could be painful when the sketch complexity is high and each character takes 2 [...]

NO! I just mean that on double click there should be shown text input field *in the center of main window*, 

This feature is NOT about "WYSIWYG" — just move behavior used in property browser for `text = {text_string} [edit]` to main window, similar to how it actually implemented for comments text editing in main window 

BUT, Additionally I still don't like how input fields "in place" text editing implemented for comments & constraint values — I guess on double click input field should be always centered in main window + text size in input field should be constant for any fields & not depend on object font/size property (as actual behavior is horrible in many situations, especially if constraints & comment uses styles in `mm` e.g. are scalable).

Input field should be fixed to system default font style & size (for example, style = "Sans", size = `14`) for ALL input fields, as it works on "Property Browser".