solvespace/solvespace #902: Add CLI option to start with custom configuration file (custom styles & workspace layout preset)

Symbian9, 2021-01-18:
### System information

- **SolveSpace version:** `3.0` (for release)
- **Operating system:** All supported

### Expected behavior

Add option to start SolveSpace with custom `*.json` (settings configuration) style.

- for GUI: `$ solvespace --conf {path_to_JSON_file}`
- for CLI: `$ solvespace-cli --conf {path_to_JSON_file}`

NOTE: If SolveSpace would start with that option, it should anyway read "Recent Files" list from default configuration file (e.g. `~/config/solvespace/settings.json`)

Implementing this feature SolveSpace users would get ability to create few presets and switch between them (without need to manually edit each style & resize windows) on star.

Each peset may include:

- custom color styles:
  - `white-black.json`, `bluepring.json`, `paper-wysivyg.json`, etc. 
- custom main window size/layout ("workspace"):
  - "fullscreen", "720x720", "half screen", etc. 
- custom tolerances & export preferences, etc.

Creating new configuration file would be simple:

1. Launch SolveSpace;
2. Set all styles & other properties for new preset (+ resize & change window position; optionally);
3. Save SLVS file & close SolveSpace;
4. Go to `~/config/solvespace/` & copy `settings.json` file;
5. Paste renamed `settings.json` as `my-new-preset.json` in same folder (or any other folder);
6. Start new SolveSpace session from terminal as `$ solvespace --conf ~/my-new-preset.json`.

### Actual behavior

Actually SolveSpace may use only `~/config/solvespace/settings.json` & overwrites it each time when on each "Save" action.

### Additional information

robnee, 2021-01-20:
I would find something like this useful.  On windows I haven't been able to find a documented way to set the default config.  I've manage to tweak the registry (so much fun converting doubles to QWORD) to fix constraint text height so I can read it but there should be a more straightforward way, hopefully one that doesn't involve the registry on Windows.

robnee, 2021-01-29:
@phkahler To start, could the json implementation of class Settings from gtk be moved to a common location for use by all three platforms as well as solvespace-cli?  Then the code from SolveSpaceUI::Init() that thaws all of the config options could be moved to an API outside SolveSpaceUI that platform/entrycli.cpp could use.  Then entry-cli.cpp could get things like ExportChordTolerance from a json Settings object rather than from command line options like --chord-tol.  An option to export the config as json would also be easy to implement across all three platforms.

If you think that's not completely misguided then I could take a look.,

phkahler, 2021-01-29:
@robnee  I have no idea how settings are saved. On Windows SolveSpace is a single executable with no installer, so either it uses a known good place for a settings file or it uses the registry. If I had to guess I'd guess file. If so, having common code for that probably makes sense. We are GPLv3, so ripping it out of GTK should be OK ;-)

I will say this though. AFAIK there is no command line options parsing in the GUI version. It seems stupid to me that there are two programs at all, but I'm sure there was at least one valid reason for it. Before adding command line options to the GUI, I'd suggest looking at the possibility of doing so by merging the CLI verison. On the surface that doesn't look hard, but it depends what the reason for it was.

robnee, 2021-01-30:
I took a look at this.  Each platform Impl stores things someplace appropriate for the platform.  On Windows its the registry and on Linux it's a JSON file in $HOME.  I thought it would be easy to reuse the JSON implementation from Linux to make an "Export Configuration" option (that would be cross platform) to easily import/export settings from a JSON file but it's a little trickier than that.

kk6mrp, 2021-01-30:
@phkahler The only request for a CLI I found was in 2014: https://solvespace.com/forum.pl?action=viewthread&parent=266 Then implemented in 2016: https://github.com/solvespace/solvespace/commit/47244c5e8988fac6526d99bdd8c5a75d669e983a No explanation for why it is a separate application. Having the GUI version able to parse command line options intuitively makes much more sense, especially in Windows :)

Symbian9, 2021-01-30:
> Before adding command line options to the GUI, I'd suggest looking at the possibility of doing so by merging the CLI verison.

@phkahler, good point!

> It seems stupid to me that there are two programs at all, but I'm sure there was at least one valid reason for it.

While merging CLI features into main GUI executable is expected as next step, there should be still an option to compile `solvespace-cli` separately as ***"headless SolveSpace"*** version (Does actual `solvespace-cli` has NO dependency to GTK? If so, that is a reason why `solvespace-cli`/"headless SolveSpace" previously was set to be separate binary, used by ex-maintainer on server for batch processing SLVS files).

phkahler, 2021-01-30:
I guess it's up to whomever implements it. The GUI version would still be built against whichever platform GUI is used (Windows, macOS, or GTK) but wouldn't display a window when handling commands. If we do this, there should be no need for a headless solvespace.

robnee, 2021-01-30:
If the goal here is to add command line parsing to all 3 platform GUIs that should be easy.  It would probably mean something like moving entrycli.cpp ShowUsage and RunCommand somewhere like src so it can be available to each platform (suggestions appreciated).  If the GUI is called with no arguments it would hop to InitGUI.  If there are command line args it would pass them to RunCommand.  I'm happy to take a look at that.  This is 4.0, right?

I would leave the question of whether to drop `solvespace-cli`for another time.

phkahler, 2021-01-30:
@vespakoen can you comment on what @robnee just asked? I'm under the impression that files in "platform" are selectively compiled depending on platform, and that entrygui.cpp and guinone.cpp are only for the cli. Is that correct?

If that's the case, I think entrygui.cpp is used on all platforms for the GUI, so we should keep that (maybe where it is for now?) and move the big functionality of entrycli into a file in /src or even bring it into entrygui since the passed options would determine if a window is opened or not. Does that make sense?

One issue is that Windows won't send standard out or err to the console from a GUI app, but google tells me it shouldn't be too hard to make it behave "correctly".

robnee, 2021-01-30:
I took a look and entrycli.cpp/solvespace-cli have a dependency on CairoPixmapRender to create png renderings.  So CAIRO_LIBRARIES would need to be linked with everything, including the GUIs.  I don't know if that is undesirable for some reason.

Symbian9, 2021-01-30:
> If we do this, there should be no need for a headless solvespace.
>
> I would leave the question of whether to drop `solvespace-cli` for another time.

To be clear here, as for now *SolveSpace* package (e.g. DEB) for Linux include  both binary: `solvespace` and `solvespace-cli`. 

If there all CLI features would be implemented for `solvespace` (GUI) then there would be no needs to include additional `solvespace-cli` binary into distribution packages.

BUT, as for me, I highly recommends to add some compiling argument to be able to compile & package `solvespace-cli` ("headless SolveSpace" with less dependency) *as separate package* for any needs to use SolveSpace for processing SLVS files on remote server (for example, to generate previews for online gallery directly on server just after new SLVS files would be uploaded to server).

Symbian9, 2021-01-30:
> One issue is that Windows won't send standard out or err to the console from a GUI app, but google tells me it shouldn't be too hard to make it behave "correctly".

I may be wrong, but while *SolveSpace* would be executed with CLI arguments via *PowerShell* (instead of `CMD.exe`) on Windows it should performs same way as it performs via *Bash* on Linux:

- https://developers.google.com/web/shows/ttt/series-2/windows-commandline

ruevs, 2021-02-01:
> I would find something like this useful. On windows I haven't been able to find a documented way to set the default config. I've manage to tweak the registry (so much fun converting doubles to [...]

@robnee Of course there is a "more straightforward way", @jwesthues is brilliant in UI design in my opinion and SolveSpace shows it (but you need to click around a bit to discover the features):
[image]
[image]
[image]

I home you see the "load factory defaults" there as well.

ruevs, 2021-02-01:
Or read the "Line Styles" section in the reference https://solvespace.com/ref.pl

ruevs, 2021-02-01:
On the subject of solvespace-cli.exe vs. eolvespace.exe supporting the command line options:
As @Symbian9 explained it is not stupid at all to have the CLI as a separate program. Very few people would ever use it and those who do would certainly know what they are doing and having minimal dependencies is beneficial for them (e.g. thumbnails or [Three.js previews](https://m-labs.hk/software/solvespace/) on a server).

On the other hand 99.99% of GUI users will never even dream that a command line exists and command line options can be passed to GUI applications, so this is mostly not needed. Inconveniences like the one that @robnee met should be solved in the GUI (if they are not already).

Finally on the specific topic of this issue: If there is going to be a way to load "configuration presets" (e.g. as .json files) it should be a GUI option.

Symbian9, 2021-02-01:
> On the other hand 99.99% of GUI users will never even dream that a command line exists and command line options can be passed to GUI applications, so this is mostly not needed.

To be clear here, main `solvespace` (GUI version) should support all CLI options of `solvespace-cli`.

> Finally on the specific topic of this issue: If there is going to be a way to load "configuration presets" (e.g. as .json files) it should be a GUI option.

Agree.

The only addition:

- for SolveSpace CLI there should be a CLI option to load configuration as `solvespace-cli --conf ~/my-preset.json [other options]`;
- for the SolveSpace GUI version there should be "configuration presets" loading button in "Property Browser" + same CLI option as `solvespace-cli` should has.

ruevs, 2021-02-01:
> > On the other hand 99.99% of GUI users will never even dream that a command line exists and command line options can be passed to GUI applications, so this is mostly not needed.
> 
> To be clear here, main `solvespace` (GUI version) should support all CLI options of `solvespace-cli`.

No, I think it should **_not_**.

Symbian9, 2021-02-01:
> No, I think it should not. Unnecessary complication, likely to be used by very few people.

It is common behavior where GUI apps has all CLI options (see *FreeCAD*, *OpenSCAD*, *Inkscape*, *GIMP*, etc).

I not see any reason why SolveSpace GUI should not support CLI options.

robnee, 2021-02-01:
> @robnee Of course there is a "more straightforward way", @jwesthues is brilliant in UI design in my opinion and SolveSpace shows it.

Agreed and thanks for the help.  Of course it works now and I have no idea now why I thought it didn't work.

robnee, 2021-02-01:
> No, I think it should not. Unnecessary complication, likely to be used by very few people.

I have a PR for this that I need to tweak to fix a build issue minor build issue.  That said I'm not sure it's appropriate. There is a clear need to retain the solvespace-cli IMHO.  It's clean and implements a separate rendering path based only on Cairo.  But It's not simply a matter of adding the existing command line parsing to the GUIs to enable this.  It requires also pulling the entire Cairo rendering path into the GUI to make it work right now.  That requires careful consideration in my opinion and isn't a slam dunk.

phkahler, 2021-02-02:
>  It requires also pulling the entire Cairo rendering path into the GUI to make it work right now. That requires careful consideration in my opinion and isn't a slam dunk.

I agree unless it's already linked to the GUI. How do we export pdf for example?

robnee, 2021-02-02:
PDF (eps/svn/etc) uses the vector renderer vs. the surface/Cairo renderer.  It does not appear that it tries to handle image requests either but I haven't had a chance to confirm that.  The vector renderer has no additional library dependencies so it's present in all builds already.  It's just transcoding the line, point, bezier representation into the target format equivalents rather than actually rendering to a pixmap. Rendering to a pixmap is done by GL in the GUI and implemented (mostly) via Cairo in the cli.

ruevs, 2021-02-02:
The GUI does **not** link to Cairo. Nor does it compile and link  `render2d.cpp` and `rendercairo.cpp` - it does not need them.

On Windows it links:
dxfrw
freetype
mimalloc-static
png_static
resources
solvespace-core
ZERO-CHECK
zlibstatic

robnee, 2021-02-02:
I'm going to close my PR related to this issue.  I think we are all in agreement that this shouldn't be changed.  I may still look at adding Json config file support to cli but even that might not be worth the effort because only a couple of config items like bgcolor and chord tolerance would even be used and there are already command line args for this.

yukulele, 2022-06-03:
minimal usual cli options should be also added:
```
-v   --version
-h   --help         
```