solvespace/solvespace #903: Error: Couldn't read file from  '--version'

Symbian9, 2021-01-19:
### System information

- **SolveSpace version:** `3.0~` https://github.com/solvespace/solvespace/commit/bd41485ebecce360baa729567273714caac1df6c
- **Operating system:** Debian 10 buster (MX Linux MX-19.3)

### Expected behavior

`solvespace` and `solvespace-cli` should has an option to show version in terminal output:

```
$ solvespace-cli --version
3.0~{git_commit_id}
$ solvespace --version
3.0~{git_commit_id}
```

Additionally `solvespace` should has `--help` as next:

```
$ solvespace --help
This is SolveSpace 3.0~{git_commit_id}

Usage: solvespace [filename...]

To use SolveSpace in CLI mode see `solvespace-cli --help`
```

### Actual behavior

Actually:

- `solvespace-cli` missed `--version` option;
- `solvespace` missed `--version` and `--help`,  instead it recognizes those option *as file names to open — as result it starts in GUI mode with error message window *"Error: Couldn't read file from  '--version'"*.

### Additional information

phkahler, 2021-01-19:
This is not relevant. Please don't create an endless list of open issues for stuff like this.

closed 2021-01-19 by phkahler

Symbian9, 2021-01-19:
> This is not relevant.

It is relevant, because actually there are NO any way to copy version number of installed app via "About" dialog (where text is not selectable by mouse/cursor!) — as result user should manually write version reading it from "About" dialog.

Also, `--version` is common CLI option which should exists at least for `solvespace-cli`.

PR #910 (2021-01-26): Add version command to solvespace-cli
Closes #903