solvespace/solvespace #904: Export 2D View as SVG does not accurately export dimension constraints of model

DeanCording, 2021-01-21:
### System information

- **SolveSpace version:** 3.0~222c80e4
- **Operating system:** Kubuntu 20.10

### Expected behavior

Exported SVG 2D view should render dimension constraints the same as displayed in Solvespace when opened in another SVG application such as Inkscape.

[image]

### Actual behavior
Dimension constraints are rendered as a mix of styles.

[image]

Both labels and lines of dimension constraints are displayed using dashed lines.  As a result, the text of the labels is very hard to read.  However, some characters in some text labels are rendered using the correct style.

### Additional information

Looking at the actual SVG data, the structure of the SVG file is a complete mess.  Some elements as have their own section but most are just lumped into a single path.

My use case for this is to use Solvespace to produce diagrams for parts that I can export as SVG and import them into our standard manufacturing documents.  Not only is the exported SVG unusable as exported, its lack of proper structure makes it impossible to edit in an SVG editor link Inkspace.

ruevs, 2021-01-21:
Related issues:
https://github.com/solvespace/solvespace/issues/518
https://github.com/solvespace/solvespace/issues/541

robnee, 2021-02-28:
> Looking at the actual SVG data, the structure of the SVG file is a complete mess. Some elements as have their own section but most are just lumped into a single path.

The code was incorrectly assigning styles to sets of closed loops instead of each loop individually producing the very long paths with everything styled as the first loop in the set.

closed 2021-02-28 by phkahler

ruevs, 2021-03-01:
I can confirm. #518 and #541 are fixed.

PR #946 (2021-02-28): Fix incorrect styling on SVG export
This PR fixes #904 (and maybe #518).  There were two problems:

1. There was no definition for Style::NO_STYLE in the header
2. The styling was being set at the SBezierLoopSet level vs the SBezierLoop so loops in the set were all being styled the same.