solvespace/solvespace #905: Can't thumbnail one of the test files?

rpavlik, 2021-01-21:
### System information

- **SolveSpace version:** 3.0 RC2
- **Operating system:** Debian testing

### Expected behavior

I should be able to do solvespace-cli thumbnail for all the slvs files I can get my hands on.

### Actual behavior

```
solvespace-cli thumbnail ./test/request/image/normal.slvs --output /%_thumb.png --size 48x48 --view isometric ./test/request/image/normal.slvs
Missing (absent) translation for group-name'#references'
Missing (absent) translation for group-name'sketch-in-plane'
File ./src/render/render2d.cpp, line 193, function DrawPixmap:
Assertion failed: false.
Message: Not implemented.
Aborted
```

Found when trying to add an autopkgtest for debian

phkahler, 2021-01-21:
Can a .slvs file contain an image? Looks like the thumbnailer is just aborting on that.

Can you post the offending file?

ruevs, 2021-01-21:
> Can a .slvs file contain an image? 

Yes. Here is a recent one https://github.com/solvespace/solvespace/issues/892#issuecomment-761884153.

First implemented here: https://github.com/solvespace/solvespace/commit/5744d1d599dfaeaa768e1667509eee1fda3b8aea

rpavlik, 2021-01-21:
The offending file is in git: test/request/image/normal.slvs

robnee, 2021-01-31:
This is coincidentally related to something I was just looking at.  The CairoRenderer, which is only used by the cli to render PNG files, is not fully implemented.  With the GUI, screenshots are exported by leveraging the GL renderer to generate the Pixmap data thus providing Image support.

I have zero "GL-fu" in my repertoire but it looks like DrawPixmap needs to be implemented for CairoRenderer or, maybe, it's possible to leverage the GL renderer in a CLI context.  If the setup of the renderer was separate from the initialization of the GUI, we might be able to reuse that for CLI rendering.  This would eliminate the dependency on the CairoRenderer and it's separate code paths.

phkahler, 2021-01-31:
Cairo might be involved in more than png export. We have pdf for example. It might not be great to use openGL for the CLI. It's one thing to link GTK and not use it. it's another to actually use OpenGL on a headless server ;-) But this stuff is outside my territory, so I'm just pointing at possible concerns and not saying what to do or not do.

robnee, 2021-01-31:
Yeah, I think you're right. The SurfaceRenderer and CairoRenderer are actually quite cool.  It's doing all of the compositing, line removal etc. that GL does but in software.  It might be the justification for leaving the stand-alone CLI alone as it doesn't require GL support.  Cairo and its dependencies are what you need to do the rendering without a GPU.  I was viewing it as a necessary evil but it's not at all, it's a very nice feature.

There's a test that mentions the fact that CairoRenderer doesn't do images so the check image test is commented out.  The test harness is using the CairoRenderer too.

Cairo is 2d but it can apply a transform matrix to the image data to shear/skew it.  I'm really out of my depth now though.

Maybe the issue should be renamed?  Less of a bug more of a feature request?

rpavlik, 2021-01-31:
Well, this came about when I was setting up "autopkgtest" - CI tests for the package on Linux. I literally just told solvespace-cli to convert every .slvs file in the test directory into every variety of output. It mostly all worked: couldn't export a surface from the ones that are just 2D (reasonable enough) and errored out trying to "thumbnail" the models with images, hitting a "not implemented". I'm not sure if anyone is using the thumbnail functionality to actually thumbnail stuff, but that seems like less-than-optimal behavior: might be better to just warn and skip trying to render the pixmap inside of a thumbnail than to error out.

robnee, 2021-01-31:
There's a test case for image requests that acknowledges what you've found.  In fact I see it's the same test you got the test file from.

```
TEST_CASE(normal_roundtrip) {
    CHECK_LOAD("normal.slvs");
    // Can't render images through cairo for now.
    // CHECK_RENDER("normal.png");
    CHECK_SAVE("normal.slvs");
}
```
I can see the value in making this degrade more gracefully.  Perhaps this should produce a console message noting the lack of image support and continue rendering. Thoughts?

phkahler, 2021-02-01:
> I can see the value in making this degrade more gracefully. Perhaps this should produce a console message noting the lack of image support and continue rendering. Thoughts?

Yes, we should not crash - ever. A console message indicating that the image is not supported would be fine for now. If anyone want's to figure out how to add the missing piece that would be welcome too. @rpavlik is this a crash report or an enhancement request? :-)

I fairly certain the CLI has not seen much use. Not too long ago we fixed the thumbnail generation for several of the orientations, so it was clear those were not being used. Some options are, as we have at least one big fan of the CLI.

ruevs, 2021-02-02:
It is very easy not to crash. I will fix it.
https://github.com/solvespace/solvespace/blob/11a8a0abd5e21a0da6c818674f8b4b657a401a6c/src/render/render2d.cpp#L193
https://github.com/solvespace/solvespace/blob/11a8a0abd5e21a0da6c818674f8b4b657a401a6c/src/render/render2d.cpp#L197

By the way `render2d.cpp` and `rendercairo.cpp` are **_not_** part of the GUI build.

ruevs, 2021-02-02:
On the other hand it would be very hard to implement. It seems to me the two OpenGL renderers use texture mapping to display a linked picture arbitrarily rotated and scaled:
https://github.com/solvespace/solvespace/blob/11a8a0abd5e21a0da6c818674f8b4b657a401a6c/src/render/rendergl1.cpp#L677
https://github.com/solvespace/solvespace/blob/11a8a0abd5e21a0da6c818674f8b4b657a401a6c/src/render/rendergl3.cpp#L544

robnee, 2021-02-02:
It looked like it would be hard to implement with 2d Cairo.I imagined image support was intended as an aid for tracing complex shapes.  Import an image with an outline of an object, trace it, probably with Bézier curves and then discard the image guide.  Images don't look like they get exported in vector formats or meshes.  Maybe exporting images in all cases is not really even very meaningful?

phkahler, 2021-04-03:
Did this get "fixed" so that doesn't crash?

ruevs, 2021-04-03:
Forgot to push my branch :-( But @robnee did :-)

PR #994 (2021-04-03): Skip image rendering in CLI rather than hard abort
This closes #905 by skipping rendering image when using CLI rather than abort since this would be difficult to implement in Cairo.  Add back disabled test cases.
closed 2021-04-03 by ruevs