solvespace/solvespace #907: Boolean fail with no boolean

phkahler, 2021-01-23:
### System information

- **SolveSpace version:** 3.0~0404cd46
- **Operating system:** Fedora

### Expected behavior

Step Translating should not have issues when the copies do not intersect/overlap

### Actual behavior

There are incomplete or missing edges/surfaces when the last group is repeated in proximity to the original.

When the translate spacing (see distance in the center) is increased the number of failures decreases. This is very strange, as there should be no surface intersections at all in this design.

### Additional information

[non_boolean_fail.zip](https://github.com/solvespace/solvespace/files/5860900/non_boolean_fail.zip)

ruevs, 2021-01-27:
That's baaad :-(

ruevs, 2021-02-01:
https://solvespace.com/forum.pl?action=viewthread&parent=3466&tt=1611846175