solvespace/solvespace #911: Add STEP (.STP) Import

aaulia, 2021-01-27:
Add ability to import STEP (.STP) file to SolveSpace.

ruevs, 2021-01-27:
Very unlikely to happen.

phkahler, 2021-01-27:
Duplicate of #88

phkahler, 2021-01-27:
@aaulia the biggest problem with STEP is reading the format. It's a huge specification, although we could probably discard a lot of it.

aaulia, 2021-01-27:
> @aaulia the biggest problem with STEP is reading the format. It's a huge specification, although we could probably discard a lot of it.

Is there any plan to implement this at all? I mean, I respect the developer decision one way or another. I will still use Solvespace whenever I can. But its inability to work with other format really limit what it can be used for. 

> Duplicate of #88  

`2016`, wow hahaha. Ok so this issue have been talked about for a long time. I understand.

closed 2021-01-27 by aaulia

phkahler, 2021-01-27:
@aaulia There isn't a plan to implement it, but we would like the ability to at least "link" step files into an assembly.  SolveSpace currently doesn't support direct editing of NURBS surfaces (a lot of people don't realize that), but linking step as a fixed part would still be valuable.

aaulia, 2021-01-27:
> @aaulia There isn't a plan to implement it, but we would like the ability to at least "link" step files into an assembly. SolveSpace currently doesn't support direct editing of NURBS surfaces (a lot of people don't realize that), but linking step as a fixed part would still be valuable.

I agree, that would still be very valuable.