solvespace/solvespace #917: Crash after trying to paste the mirror of a construction

someSven, 2021-01-31:
### System information

- **SolveSpace version:3.0~3e3ccdca
- **Operating system:Raspian

### Expected behavior
Pasting the mirror of a construction. Only some lines, including a bezier line, and some constraints. 

### Actual behavior
Programms crashed and window disappeared.

### Additional information
Shell:
Generate::ALL (for bounding box) took 90 ms
Generate::ALL took 91 ms
File ./src/dsc.h, line 479, function FindById:
Assertion failed: t != NULL.
Message: Cannot find handle.
Aborted

[thighV2_front-outer.slvs.zip](https://github.com/solvespace/solvespace/files/5899558/thighV2_front-outer.slvs.zip)

phkahler, 2021-01-31:
It would be helpful to know which group and which entites you tried to paste so we can reproduce the crash. I was able to copy and paste-transformed the entire g002 sketch with the "flipped" checkbox so it was reversed. Not sure copy paste is really a thing in a 3D group, but if that's what you tried to do it should at least not crash...

ruevs, 2021-02-01:
I can not cause it to crash either. I even did Sketch | In Workplane in group g003-revolve, Ctrl-A, Ctrl-C, Ctrl-T and it still works (does stupid things of course, but does not crash).

someSven, 2021-02-01:
Ah, I cut it, instead of copying. Because I didn't need the old construction, just wanted it to be on the other side. With copy it works, even into a new workplane.

ruevs, 2021-02-01:
Aha with cut -> paste transformed it crashes!
Ufff this is tricky...
https://github.com/solvespace/solvespace/blob/11a8a0abd5e21a0da6c818674f8b4b657a401a6c/src/clipboard.cpp#L171

vespakoen, 2021-02-03:
That looks tricky indeed, maybe a good solution is to disable paste transformed when the selection was cut and add a "Transform" option that works like paste transformed but does a copy itself and deletes the original selection when it's done.

phkahler, 2021-05-16:
Related to #808.