solvespace/solvespace #918: Crash after autosave-restore dialog wth file from v2.3

someSven, 2021-01-31:
### System information

- **SolveSpace version:3.0~3e3ccdca
- **Operating system:Raspian wth OpenGL active

### Expected behavior
Autosave file from v2.3 should open on v3 when I click to load it in the startup dialog.

### Actual behavior
Crash.

### Additional information
- I have OpenGL activated.
- The file was created in Solvespace 2.3, it works if I choose not to restore the autosave. 
- It's not important to me, just thought you should know.

Shell:
(solvespace:18054): Gtk-CRITICAL **: 10:57:23.745: gtk_file_chooser_widget_set_current_name: assertion 'priv->action == GTK_FILE_CHOOSER_ACTION_SAVE || priv->action == GTK_FILE_CHOOSER_ACTION_CREATE_FOLDER' failed
File ./src/graphicswin.cpp, line 394, function Init:
Assertion failed: !SK.groupOrder.IsEmpty().
Message: Group order can't be empty since we will activate the last group..
Aborted

[Chuwi-Guard.slvs.zip](https://github.com/solvespace/solvespace/files/5899573/Chuwi-Guard.slvs.zip)

phkahler, 2021-01-31:
Not sure what the issue is, but it looks like a problem in 2.3 or even current version with auto-save. There are 2 files in the zip archive. One of them is zero length so I opened the other one and it works fine. Is the problem actually the empty file? Does it actually work on 2.3?
[image]

someSven, 2021-02-01:
I tried again. The problem is the empty autosave file, which v2.3 created at some point. After putting it back into the same dir than the file I want to open, it crashes after trying to load the real file, if I click on "Load autosave". Same with both versions of Solvespace. If I click not to load it, then it works with both versions, the same if I delete the autosave-file.

ruevs, 2021-02-01:
There is a problem when opening empty (0 bytes long) .SLVS files.
I'll fix it.

[image]

The problem is that here:
https://github.com/solvespace/solvespace/blob/11a8a0abd5e21a0da6c818674f8b4b657a401a6c/src/file.cpp#L480
we clear all groups. Then we try to read an empty file and therfore this `while` exists immediately:
https://github.com/solvespace/solvespace/blob/11a8a0abd5e21a0da6c818674f8b4b657a401a6c/src/file.cpp#L487
and we come to here:
https://github.com/solvespace/solvespace/blob/11a8a0abd5e21a0da6c818674f8b4b657a401a6c/src/file.cpp#L548
where the is no `fileLoadError` and thus we are left with no groups, and so this assert fails:
https://github.com/solvespace/solvespace/blob/11a8a0abd5e21a0da6c818674f8b4b657a401a6c/src/graphicswin.cpp#L394
since is is not allowed to have no groups :-)

ruevs, 2021-02-01:
Fixed.

closed 2021-02-01 by someSven