solvespace/solvespace #920: "dof" in group list lags by "one click" after applying a constraint with the keyboard.

rpavlik, 2021-02-03:
### System information

- **SolveSpace version:** 3.0~11a8a0ab (openmp edge build)

- **Operating system:** Win 10 2004

### Expected behavior
As soon as I hit `O` to constrain the point on the plane, it should have flipped back to the groups view and shown 1DOF.

### Actual behavior

It flipped back and showed 2 dof until I selected something then un-selected it.

### Additional information

[image]

You can do the same thing with just a rectangle, it turns out. If you click the "symmetric" toolbar button, you'll actually see the 4 turn into a 3 after it goes back to the groups screen. (Think my screen capture was too slow to see it though) If you do the keyboard, it won't go from 3 to 2 until you force something to happen. (You'll just have to imagine, based on the property browser, when I actually pressed `y` in this capture. when it shows down to 2, that's from  me just clicking) Really weird.

[image]

ruevs, 2021-02-03:
Indeed. Probably related to this one: https://github.com/solvespace/solvespace/issues/863

rpavlik, 2021-02-03:
oh, interesting. This feels like a regression to me, I noticed it pretty quick and it seemed surprising. I have "edit newly added" enabled so presumably that's why I didn't notice it with the dimension constraint. Wonder if we "recently" (I haven't done much modeling in the last 2 months) removed one forced-recalc...

ruevs, 2021-02-04:
We did remove one on very recently 2021-01-31 https://github.com/solvespace/solvespace/commit/11a8a0abd5e21a0da6c818674f8b4b657a401a6c https://github.com/solvespace/solvespace/pull/915 but this behaviour is the same before it.

I have to bisect to be sure,  but this may be as it is since 8192c965eaa when it was implemented (https://github.com/solvespace/solvespace/pull/290 https://github.com/solvespace/solvespace/pull/200)
and #863 (the `err` one) could be like this for a long time, it was last touched here c2c26e95adb.

Another relevant commit 9148d0cb9

rpavlik, 2021-02-04:
oh, thanks. This is not a big deal, I just wanted to record it while I still had a reproducer open.

robnee, 2021-02-07:
I think this is the same timer ordering bug as noticed when working on #863 so I think I can fix this too.

PR #1257 (2022-06-14): Make scheduled/deferred task order deterministic
Fixes #920, #1143

Explanation from @robnee on Feb 7, 2021 in pull request #927

Solvespace uses two timers (generateAllTimer and showTWTimer) to defer tasks until the event loop processing finishes. This helps coalesce multiple calls into one. You can call scheduleGenerateAll multiple times while processing UI messages but only trigger one GenerateAll. scheduleGenerateAll and scheduleShowTW do their scheuduling by setting timers with durations of zero. These timers fire (at least on Linux and Windows) some time after all other events in the mes [... cut]
commit b48ce240 (2022-06-14): Make scheduled/deferred task order deterministic

Fixes #920 #1143

Explanation from @robnee on Feb 7, 2021 in pull request #927

Solvespace uses two timers (generateAllTimer and showTWTimer) to defer tasks
until the event loop processing finishes. This helps coalesce multiple calls
into one. You can call scheduleGenerateAll multiple times while processing UI
messages but only trigger one GenerateAll. scheduleGenerateAll and
scheduleShowTW do their scheuduling by setting timers with durations of zero.
These timers fire (at least on Linux and Windows) some time after all other
events in the mes [... cut]
closed 2022-06-14 by phkahler