solvespace/solvespace #921: "Variadic" constrain equal

rpavlik, 2021-02-03:
This is a feature requests for post-3.0. It would be really slick if I could select more than 2 items and hit q to constrain them all equal. This is an operation I wish I had quite frequently, as I'll often need repeated features that I don't step/repeat, etc. but want same size. Sometimes I'll even go ahead and select them all and try it, before realizing that oops, yeah that's an error.

ruevs, 2021-02-03:
There are two pull requests https://github.com/solvespace/solvespace/pull/530 https://github.com/solvespace/solvespace/pull/836
And an issues https://github.com/solvespace/solvespace/issues/573 about this already ;-)

Obviously a popular idea, definitely post 3.0

phkahler, 2021-02-03:
Very soon after 3.0 so we have lots of time to get it right. But remember we try not to break existing sketches, which can happen while tweaking implementation details.

I really want this feature too.

rpavlik, 2021-02-03:
oops, sorry for the dupe, I can close this if it's well captured elsewhere. I would just implement it as UI, constrain all to be equal to the first, but that's just me.

closed 2022-12-31 by phkahler