solvespace/solvespace #928: Undo symmetric constraint does not restore auto-removed H or V constraints

robnee, 2021-02-07:
### System information

- **SolveSpace version:** 3.0~02aa9ea8
- **Operating system:** Windows 10

### Expected behavior

Draw a rectangle about the origin and constrain one of the sides symmetric.  This will automatically remove a redundant H or V constraint.  Hit C-z to undo the symmetric constraint and restore the auto-removed H or V constraint.

### Actual behavior

While Undo removes the symmetric constraint it does not restore the auto-removed H or V.

### Additional information

@phkahler I've started working on a fix by adding an `UndoRemember()` call before auto-removing the H or V constraint.  This works but requires two undos to fully restore the sketch.  I suppose this is an improvement but is there a way to collapse this into a single undo stack entry so only one undo is required?

robnee, 2021-02-07:
There is a similar problem with the midpoint constraint if the point is already constrained on the line.  The on-line constraint will be auto-removed but undo will not restore.

phkahler, 2021-02-07:
I did a test:

1) Draw rectangle.
2) Draw some random line segment.
3) Constrain one horizontal line of rectangle Y.
4) Constrain one vertical line of rectangle Y.
5) Undo
6) Undo
7) Undo

At step 5, neither H or V constraint reappears.
At step 6, the V constraint removed in step 4 reappears.
At step 7 the H constraint reappears while the line is removed.

Does this mean the constraint deletions are there, but in the wrong order in the history?

robnee, 2021-02-07:
Yes.  `UndoRemember()` isn't being calling at the right times.  It's called between the time the redundant constraint is removed and the new one is added.  The PR changes this to the add/remove calls are grouped together in the Undo stack.

PR #929 (2021-02-16): remember auto-removed constraints for undo
This PR adds calls to `UndoRemember` before auto-removing constraints when adding midpoint or symmetric constraints.  The subsequent call to `AddConstraint` is changed to `rememberForUndo=false` in order to prevent requiring the user to undo twice to return to the starting state.

Fixes #928
closed 2021-02-16 by phkahler