solvespace/solvespace #930: Move Axis icon in the down right corner

Symbian9, 2021-02-08:
### System information

- **SolveSpace version:** `3.0-git` https://github.com/solvespace/solvespace/commit/02aa9ea827a128636e2d56d472e25b311868e073
- **Operating system:** Debian 10 buster

### Expected behavior

Move axis icon displayed in main window from "down left"  to "down right" position.

[image] [image]

Alternatively add it as a togglable option on *"Property Browser > configuration"* tab.

### Actual behavior

Actually axis icon is placed in "down left" corner, but on smaller screens it starts jump in far right position from toolbox panel.

Keepin toolbar & axis icon on the same side in UI decreases readability of interface.

[pic.2](https://i.imgur.com/6EwHFS2.png) [pic.3](https://i.imgur.com/1dv3eDt.png)

### Additional information

phkahler, 2021-02-15:
Should this be on the right side only if it does not fit under the toolbar? It seems like always putting it on the right would clutter the drawing area.

ruevs, 2021-02-15:
Definitely NO from me to this change. Exactly on small screens I may put the text window over the axis and tool bar (mostly unused due to keyboard shortcuts and menus).

And another check box for this is also completely useless in my opinion - I do not like software what has many options that do not really do anything useful but clutter the interface and make it harder to find the good stuff.

Symbian9, 2021-02-17:
> Should this be on the right side only if it does not fit under the toolbar?

I suggest there should be an option (in configuration screen) to set on which side axis icon should be placed: on left side OR on right side.

> Definitely NO from **me** to this change. Exactly on small screens **I may** put the text window over the axis and tool bar (mostly unused due to keyboard shortcuts and menus).

That is why I propose give users control over UI layout, so they may set & customize position of axis icon according own needs.

ruevs, 2022-10-09:
@phkahler I'm closing this to reduce noise.

closed 2022-10-09 by ruevs