solvespace/solvespace #931: Feature request

nurchi, 2021-02-10:
### System information

SolveSpace version: 3.0~36719c62
Operating system: Windows 10 Pro 20H2

### Feature request

Named parameters
The idea here is to have named parameters on a model with some default values and possibly valid ranges.

For example, one could design a bolt specifying its diameter and thread pitch. Then the design would be affected by these values.

When making an assembly, there would be no need to make a part for each case, instead a common part (e.g. a library part) could be used, and by setting a parameter, the look of that instance would change.

Hope it's clear.

ruevs, 2021-02-11:
This is a good idea. And it has been discussed in the past.
See here https://github.com/solvespace/solvespace/issues/77

closed 2021-02-11 by ruevs

henrebotha, 2022-03-05:
This issue should be renamed so that references to it elsewhere (e.g. in #1024) have a sensible name to display instead of just "Feature request".