solvespace/solvespace #932: Poor performance when extruding translate groups

baryluk, 2021-02-11:
### System information

- solvespace from master @ 7795b0bc1911a01eaa4661e252488d8f07ec2135
- Debian testing, amd64
- gcc 10.2.1

### Details

I was doing a small project for fun, and went a bit over board with some holes in a sheet metal, to the point where it takes about 4 minutes to render do anything useful.

I used an extrusion of translate of translate to generate a 2D pattern.

[case-vents.slvs.gz](https://github.com/solvespace/solvespace/files/5962704/case-vents.slvs.gz)  (~20MB unpacked, can take few minutes to open, be warned).

Looking at `perf top`, and a bit of code in `Generate`, it looks like the major culprit it CopyEntity, which triggers a copy of Entity classes. The base class (`EntityBase`) has two `std::string` fields ``str` and `font`, which I believe should not be in the `EntityBase`, as most of the entities doesn't use these fields actually. Because they are there directly, each construction and destruction, will be called every time any entity is copied. In perf I see `~basic_string<...>`, using about 80% of the CPU.

ruevs, 2021-02-12:
Yes it is slow on the path you described...
But for me on Windows (RelWithDebInfo) most of the time is spent in `SolveSpace::Entity::operator=` on this path.

```
Function Name	Total CPU [unit, %]	Self CPU [unit, %]	Module	Category
|||| + SolveSpace::Platform::RunGui	44611 (83.83 %)	0 (0.00 %)	solvespace.exe	UI | IO | Graphics | File System | Kernel
||||| + [External Code]	44611 (83.83 %)	55 (0.10 %)	Multiple modules	UI | IO | Graphics | File System | Kernel
|||||| - std::_Func_impl_no_alloc<<lambda_8b75e19112ec9ef5254c574e001b6b54>,void>::_Do_call	26118 (49.08 %)	0 (0.00 %)	solvespace.exe	IO | Kernel
|||||| + std::_Func_impl_no_alloc<std::_Binder<std::_Unforced,void (__thiscall SolveSpace::SolveSpaceUI::*)(enum SolveSpace::SolveSpaceUI::Generate,bool,bool),SolveSpace::SolveSpaceUI *,enum SolveSpace::SolveSpaceUI::Generate,bool,bool>,void>::_Do_call	13772 (25.88 %)	0 (0.00 %)	solvespace.exe	IO | Kernel
||||||| + SolveSpace::SolveSpaceUI::GenerateAll	13772 (25.88 %)	0 (0.00 %)	solvespace.exe	IO | Kernel
|||||||| + SolveSpace::SolveSpaceUI::GenerateAll	5969 (11.22 %)	0 (0.00 %)	solvespace.exe	Kernel
||||||||| + SolveSpace::Group::Generate	5918 (11.12 %)	8 (0.02 %)	solvespace.exe	Kernel
|||||||||| + SolveSpace::Group::CopyEntity	4247 (7.98 %)	7 (0.01 %)	solvespace.exe	Kernel
||||||||||| + SolveSpace::IdList<SolveSpace::Entity,SolveSpace::hEntity>::Add	4212 (7.92 %)	2 (0.00 %)	solvespace.exe	Kernel
|||||||||||| + std::inplace_merge<SolveSpace::Entity *,SolveSpace::CompareId<SolveSpace::Entity,SolveSpace::hEntity> >	4192 (7.88 %)	3525 (6.62 %)	solvespace.exe	Kernel
||||||||||||| + std::_Rotate_one_right<SolveSpace::Entity *>	650 (1.22 %)	9 (0.02 %)	solvespace.exe	Kernel
|||||||||||||| + SolveSpace::Entity::operator=	639 (1.20 %)	634 (1.19 %)	solvespace.exe	Kernel
```

ruevs, 2021-02-12:
Here is a much simpler test case for this
[LeliaManya2.zip](https://github.com/solvespace/solvespace/files/5974674/LeliaManya2.zip)

robnee, 2021-02-13:
@ruevs if it's copying the str and font strings in EntityBase then perhaps they could be moved to a new Text class and replaced with a (new) hText?  Presumably those are just an int32 copy.

baryluk, 2021-02-14:
I don't think the move of strings would solve it completely, but it should in general help quite a bit. It is good to keep entity memory sizes as small as possible anyway, to reduce memory usage and improve CPU cache utilization. Even just few percent improvements in memory or CPU time, would be welcome.

phkahler, 2021-02-14:
You might want to have a look at the second patch mentioned in [issue #430](https://github.com/solvespace/solvespace/issues/430)

Specifically https://github.com/Evil-Spirit/solvespace-master/commit/9cf1a09bcdaa5a2bc99400d368dc1a8f43996e1d
or as an actual patch: https://github.com/Evil-Spirit/solvespace-master/commit/9cf1a09bcdaa5a2bc99400d368dc1a8f43996e1d.patch

If that helps this situation I'll take it as another data point for "we need to fix the list implementation", maybe using that patch. I would prefer that we use a better underlying data structure though.

Unfortunately it no longer applies cleanly for me (using git apply).

baryluk, 2021-02-15:
> You might want to have a look at the second patch mentioned in [issue #430](https://github.com/solvespace/solvespace/issues/430)
> 
> Specifically [Evil-Spirit@9cf1a09](https://github.com/Evil-Spirit/solvespace-master/commit/9cf1a09bcdaa5a2bc99400d368dc1a8f43996e1d)
> or as an actual patch: https://github.com/Evil-Spirit/solvespace-master/commit/9cf1a09bcdaa5a2bc99400d368dc1a8f43996e1d.patch
> 
> If that helps this situation I'll take it as another data point for "we need to fix the list implementation", maybe using that patch. I would prefer that we use a better underlying data structure though.
> 
> Unfortunately it no longer applies cleanly for me (using git apply).

Why `class List` and `class IdList` even exists? Why not `std::vector` and `std::unordered_map` or equivalent (there are some faster / smaller variants depending on application)? Just curious.

ruevs, 2021-02-15:
> Why `class List` and `class IdList` even exists? Why not `std::vector` and `std::unordered_map` or equivalent (there are some faster / smaller variants depending on application)? Just curious.

@baryluk See here:
https://github.com/solvespace/solvespace/pull/433 https://github.com/solvespace/solvespace/pull/429
@rpavlik started this a long time ago... but (I think) getting it right and not actually reducing performance was hard.

rpavlik, 2021-02-15:
Yes, getting it right and not ending up with memory leaks, use after free, or other corruption (failing test) was very tricky. Performance wasn't hard, but the semantics of the existing class is not super clear. (Shallow copy is intentional, which is why my latest attempts wrapped a shared ptr holding a standard container) It was a pathological case like this that got me started on that path years ago though.

I think that all the changes that I managed to get working right have been merged already. I can offer tips if someone wants to pick up where I left off though.

phkahler, 2021-02-15:
This rekindled some of my curiosity about Group::CopyEntity().  I was under the impression that a copy is only made if one fitting the description does not already exist. For example, if you change the number of copies in a step-translating group from 3 to 5 it should only create 2 additional copies of every entity and leave the 3 existing ones alone. This is handled by Remap which takes an entity handle and a copy-number to get or create a new handle. But looking now, it seem to be creating a new entity regardless of weather one already exists with the same handle. Is that true? If so, what does the list do when Add is called with the duplicate?

phkahler, 2021-02-19:
One common performance issue with the list class is when algorithms set one element.tag = 1 and then call l.RemoveTagged. In some places multiple items are tagged prior to that call, but for the cases that only remove a single list item we should probably implement a remove(index) function and use that, so if someone tries to make a better list implementation they will know to make remove O(1) or O(log n) instead of the current O(n). That could be a win all by itself. Come to think of it, removing a specific item could probably be done with the current implementation faster than calling RemoveTagged for one item.

rpavlik, 2021-02-19:
It's been a while since I last profiled SolveSpace. I'm not sure we really want to switch to something like a linked list that would give us O(1) deletion: skipping the tagging step would just reduce our coefficient on O(N) since we still have to move the following stuff, right? Though, if we used an STL algorithm and had a remove one function, we'd be more likely to get an optimized block move in our deletion, which wouldn't impact big O but might make a big impact in reality sure to memory access patterns

phkahler, 2021-02-19:
@rpavlik  I guess by grouping tagging a single item followed by calling DeleteTagged into one function, it would allow us to try alternative implementations. RemoveTagged can remove multiple items, so it must be O(n). If some function actually wants to delete one item, we should have a list method for that.

There are several places in the code that take a list (of edges is a good example) and end up removing them one by one using RemoveTagged. That's going to be O(n^2) until either the algorithm changes or we have a better delete.

If we tried some other structure (like a tree) instead of a list, we might have O(log n) lookup, but also O(log n) delete. We wouldn't see any improvement because we're doing RemoveTagged all the time instead of a single delete. We're often using a function that does more than is needed, so we're locked in to that time complexity.

rpavlik, 2021-02-19:
ah, good thought. hadn't thought about the "deleting one by one" locations being n2, didn't realize we had those. Presumably that type of usage is why we have the tag and delete in the first place...

Definitely in favor of improving the semantic content of the code by replacing "tag a single item and remove tagged" with a " remove a single item" primitive.

pjanx, 2021-03-27:
@baryluk I believe I've mostly solved your issue in #973, so feel free to retry on current HEAD.

pjanx, 2021-03-27:
When I increase the number of holes in either direction too much, Solvespace crashes on this, though:
```
File ./src/dsc.h, line 468, function Add:
Assertion failed: FindByIdNoOops(t->h) == nullptr.
Message: Handle isn't unique.
```
I don't know what to think of it. I've touched it last...

ruevs, 2021-03-27:
> When I increase the number of holes in either direction too much, Solvespace crashes on this, though:
> 
> ```
> File ./src/dsc.h, line 468, function Add:
> Assertion failed: FindByIdNoOops(t->h) == nullptr.
> Message: Handle isn't unique.
> ```
> 
> I don't know what to think of it. I've touched it last...
I'm writing from  a phone, so I can't be sure, but most probably more than 65536 objects/handles.

phkahler, 2021-03-27:
Redoing the handles will be very tricky to do without breaking every existing sketch. It may require a converter in the file reading (and linking) code.

But we should verify that's the problem and not a new bug.

closed 2021-05-12 by phkahler

phkahler, 2022-03-22:
I'm wondering if the issue here is the O(n**2) behavior of ear triangulation and SEdgeList::AssemblePolygon / SEdgeList::AssembleContour

PR #1002 (2021-05-12): Optimize IdList by Using an Index and a Proper Iterator
This drastically speeds up IdList operations. Closes #932 and #939 and even the remainder of #841.
It is similar to https://github.com/solvespace/solvespace/pull/992 and https://github.com/Evil-Spirit/solvespace-master/commit/44eb1a4703a1be50d74fdfeacb97f89c06e08ab6

It optimizes IdList by Using an Index and a Proper Iterator:
    
- Use `std::vector<T> elemstore` to store elements. Avoids manual memory management.
- Add an index (`std::vector<int> elemidx`) that avoids moving large objects  when adding elements.
- Add a free element list [... cut]