solvespace/solvespace #933: Different scaling factors on X,Y,Z when linking (importing)

antariandel, 2021-02-11:
### System information

- **SolveSpace version:** 3.0~36719c62
- **Operating system:** Windows 10

### Idea

This is more like an idea/question/feature request and less of an issue. I'm curious if it would be possible and reasonable to allow applying different scaling factors on the 3 othogonal axes on linked groups?

Thank you all for maintaining this piece of software by the way, I really grew fond of it.

phkahler, 2021-02-12:
FYI if you want to build by linking extrusions ( beams or 8020 ) you can create the cross section as a 2D sketch, link it, and extrude to length *after* linking. You can link the same sketch many times and have each extrusion a different length.

Your request for separate scale factors isn't likely to happen but we can leave it open.

antariandel, 2021-02-12:
Thanks for the reply! I knew of that option, but I'd like to change 2 dimensions when linking (width and depth).

As I continued to think about this, the functionality that would provide what I'm seeking, and much more would be to have the ability to assign variables to certain dimension constraints, and have the option to change the value of these variables either inside the file that defined them, but also in the assembly file when linking. Having the option to suppress some groups within the linked file (to turn some features - ie. parts - on or off) would extend the functionality even more. This would vastly widen the scope of parametric modelling in SolveSpace.

This idea comes from Onshape, which has what they call "Configurations."

I realize this is out of scope of the topic of this issue, but I'm curious if you think this is something worth pursuing.

phkahler, 2021-02-12:
OK this is a duplicate of issue #77  which we would like to do some day.

closed 2021-02-12 by phkahler