solvespace/solvespace #934: Mimalloc snuck in a -march=native on ARM

rpavlik, 2021-02-11:
### System information

- **SolveSpace version:** 3.0~rc2
- **Operating system:** e.g. Debian testing

Debian's great QA automations found a `-march=native` in the build log on arm: turns out mimalloc put that in there. Filing this issue here because I had to work around it in the Debian package for SolveSpace, but it's really a mimalloc bug. `-march=native` should be a user choice.

https://qa.debian.org/bls/packages/s/solvespace.html
[image]

ruevs, 2021-02-12:
@rpavlik It disappeared? Did the build already take this https://github.com/microsoft/mimalloc/pull/362 into account?

rpavlik, 2021-02-12:
Ah, nifty - that's why I screenshot it. Somebody kindly sponsored my package update that patched it out:
[image]

rpavlik, 2021-02-12:
That said, I totally don't understand the other two warnings on that page.

robnee, 2021-02-13:
This seems relevant to the hidden flags warning

https://manpages.debian.org/testing/blhc/blhc.1.en.html

phkahler, 2021-04-03:
@rpavlik  I finally bumped to their latest dev-slice since there's been no updates in a month.

PR #989 (2021-04-03): Update mimalloc version.
This close #947   close #934

Help with #913   - no patches required.
closed 2021-04-03 by phkahler