solvespace/solvespace #935: Segmentation fault

ardugitter, 2021-02-13:
### System information

Debian stable
3.0~02aa9ea8

### Expected behavior

Should open the file without crashing as it did on an old version of Solvespace. I am not sure what version I created the file on, but I opened it for the first time in 3 months and it crashes my system.

The file just contains a revolved pattern (360 degrees) and a couple of difference extrusions. It appears as a sort of disc shape if opening correctly.

Please rename the attached file with a .slvs extension.

### Actual behavior

Segmentation fault appears when run from console. Locks up my system.

### Additional information

Please try opening the attached slvs but be warned, it may crash your computer or at least slow it down dramatically.

ardugitter, 2021-02-13:
[test.txt](https://github.com/solvespace/solvespace/files/5975582/test.txt)

ruevs, 2021-02-14:
It does not crash for me on Windows. 
Did you compile from source? (02aa9ea8 is pretty new - from Feb 04 only two commits after it)
[image]

ardugitter, 2021-02-14:
Thanks, that is what it is meant to look like.

I built from source using the latest version. I suppose I should go back to a stable version?

Do you think my slvs file is helpful to the developers?

ruevs, 2021-02-14:
The master is as stable as it gets, so you do not need to go back when building from source.
However it may be interesting to try the pre-compiled Linux binaries (from the releases page or from Snap) just to see if they crash as well.

Also I think (not using Linux actively), that if you start it from a console it should dump some debug information when it crashes. It may be useful if you paste it here.

ardugitter, 2021-02-14:
I also tried v3rc2 and had the same problem.

I started it from the console and got this:

```
(solvespace:4341): dbind-WARNING **: 15:39:45.746: Couldn't register with accessibility bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.

(solvespace:4341): Gtk-WARNING **: 15:39:45.753: Theme parsing error: gtk.css:127:35: The style property GtkButton:child-displacement-x is deprecated and shouldn't be used anymore. It will be removed in a future version

(solvespace:4341): Gtk-WARNING **: 15:39:45.753: Theme parsing error: gtk.css:128:35: The style property GtkButton:child-displacement-y is deprecated and shouldn't be used anymore. It will be removed in a future version

(solvespace:4341): Gtk-WARNING **: 15:39:45.753: Theme parsing error: gtk.css:129:34: The style property GtkCheckButton:indicator-size is deprecated and shouldn't be used anymore. It will be removed in a future version

(solvespace:4341): Gtk-WARNING **: 15:39:45.753: Theme parsing error: gtk.css:130:36: The style property GtkCheckMenuItem:indicator-size is deprecated and shouldn't be used anymore. It will be removed in a future version

(solvespace:4341): Gtk-WARNING **: 15:39:45.753: Theme parsing error: gtk.css:132:46: The style property GtkScrolledWindow:scrollbars-within-bevel is deprecated and shouldn't be used anymore. It will be removed in a future version

(solvespace:4341): Gtk-WARNING **: 15:39:45.753: Theme parsing error: gtk.css:135:30: The style property GtkExpander:expander-size is deprecated and shouldn't be used anymore. It will be removed in a future version

(solvespace:4341): Gtk-WARNING **: 15:39:45.754: Theme parsing error: gtk.css:142:29: The style property GtkStatusbar:shadow-type is deprecated and shouldn't be used anymore. It will be removed in a future version
SolveSpace!

Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Killed
```

This is using master as you suggested, that is 3.0~7795b0bc

ruevs, 2021-02-14:
Hmm nothing interesting. It just closes? I don't know, some of the guys running it on Linux will have to check....
As for the version - you mentioned 02aa9ea827a1 in the issue above, but this is just two commits before 7795b0bc1911a01eaa4661e252488d8f07ec2135 (https://github.com/solvespace/solvespace/commits/master)...

phkahler, 2021-02-14:
It opens fine for me on Fedora 33.

ardugitter, 2021-02-14:
> It opens fine for me on Fedora 33.

Do you know exactly what version you are using please? Thanks.

phkahler, 2021-02-15:
> > It opens fine for me on Fedora 33.
> 
> Do you know exactly what version you are using please? Thanks.

Latest master built from source yesterday. BTW most of the terminal output you saw was there for me too, the difference was that it didnt crash for me.

ardugitter, 2021-02-15:
Okay, thanks. Did you build with "cmake .. -DCMAKE_BUILD_TYPE=Release -DENABLE_OPENMP=ON"?

ruevs, 2021-02-15:
@ardugitter can you debug it?

ardugitter, 2021-02-16:
I built the debug version using:

`cmake .. -DCMAKE_BUILD_TYPE=Debug`

Here is the output from gdb. Note that I did not Ctrl-C at any point.

```
Reading symbols from solvespace...done.
(gdb) run test.slvs
Starting program: /usr/local/bin/solvespace 50cm_canister_brace_5.slvs
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

(solvespace:19655): dbind-WARNING **: 15:32:35.806: Couldn't register with accessibility bus: Did not receive a reply. Possible causes include: the remote application did not send a reply, the message bus security policy blocked the reply, the reply timeout expired, or the network connection was broken.
[New Thread 0x7ffff1ac2700 (LWP 19659)]
[New Thread 0x7ffff12c1700 (LWP 19660)]

(solvespace:19655): Gtk-WARNING **: 15:32:35.834: Theme parsing error: gtk.css:127:35: The style property GtkButton:child-displacement-x is deprecated and shouldn't be used anymore. It will be removed in a future version

(solvespace:19655): Gtk-WARNING **: 15:32:35.834: Theme parsing error: gtk.css:128:35: The style property GtkButton:child-displacement-y is deprecated and shouldn't be used anymore. It will be removed in a future version

(solvespace:19655): Gtk-WARNING **: 15:32:35.834: Theme parsing error: gtk.css:129:34: The style property GtkCheckButton:indicator-size is deprecated and shouldn't be used anymore. It will be removed in a future version

(solvespace:19655): Gtk-WARNING **: 15:32:35.834: Theme parsing error: gtk.css:130:36: The style property GtkCheckMenuItem:indicator-size is deprecated and shouldn't be used anymore. It will be removed in a future version

(solvespace:19655): Gtk-WARNING **: 15:32:35.834: Theme parsing error: gtk.css:132:46: The style property GtkScrolledWindow:scrollbars-within-bevel is deprecated and shouldn't be used anymore. It will be removed in a future version

(solvespace:19655): Gtk-WARNING **: 15:32:35.835: Theme parsing error: gtk.css:135:30: The style property GtkExpander:expander-size is deprecated and shouldn't be used anymore. It will be removed in a future version

(solvespace:19655): Gtk-WARNING **: 15:32:35.835: Theme parsing error: gtk.css:142:29: The style property GtkStatusbar:shadow-type is deprecated and shouldn't be used anymore. It will be removed in a future version
SolveSpace!

[New Thread 0x7fffeac51700 (LWP 19661)]
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Generate::REGEN took 56 ms
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Generate::ALL (for bounding box) took 37 ms
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
Vector::WithMagnitude(1) of zero vector!
mimalloc: warning: mi_usable_size: pointer might not point to a valid heap region: 0x7fffe8810000
(this may still be a valid very large allocation (over 64MiB))
mimalloc: warning: (yes, the previous pointer 0x7fffe8810000 was valid after all)
mimalloc: warning: mi_usable_size: pointer might not point to a valid heap region: 0x7fffe8810000
(this may still be a valid very large allocation (over 64MiB))
mimalloc: warning: (yes, the previous pointer 0x7fffe8810000 was valid after all)
mimalloc: warning: mi_usable_size: pointer might not point to a valid heap region: 0x7fffe8810000
(this may still be a valid very large allocation (over 64MiB))
mimalloc: warning: (yes, the previous pointer 0x7fffe8810000 was valid after all)
mimalloc: warning: mi_usable_size: pointer might not point to a valid heap region: 0x7fffe8810000
(this may still be a valid very large allocation (over 64MiB))
mimalloc: warning: (yes, the previous pointer 0x7fffe8810000 was valid after all)
[Thread 0x7ffff12c1700 (LWP 19660) exited]
[Thread 0x7fffeac51700 (LWP 19661) exited]
[Thread 0x7ffff1ac2700 (LWP 19659) exited]

Program terminated with signal SIGKILL, Killed.
The program no longer exists.
```

Problem with mimalloc perhaps?

phkahler, 2021-02-16:
Oh crap. I'm sorry. You're on Debian? See #924 and #893.  But assuming you're not on ARM we shall as @rpavlik if there's still something up with master building on Debian in terms of mimalloc?

rpavlik, 2021-02-16:
What platform/architecture are you on? (x64, arm, arm64, other?)  Can you build with one of the sanitizers enabled, to see if that can find our bug?

ardugitter, 2021-02-16:
I'm on x64 and I have a weird Debian setup. That said, I am doing the usual git clone and submodule update.

Please direct me to build with a sanitizer and I will go ahead and do that. Thanks.

phkahler, 2021-02-28:
@ardugitter 
-DENABLE_SANITIZERS=ON

ardugitter, 2021-03-01:
@phkahler 

Thanks. I also passed in the Debug flag to cmake.

Now solvespace does not run at all:

```
$ solvespace

(solvespace:6659): Gtk-WARNING **: 12:10:55.764: Theme parsing error: gtk.css:127:35: The style property GtkButton:child-displacement-x is deprecated and shouldn't be used anymore. It will be removed in a future version

(solvespace:6659): Gtk-WARNING **: 12:10:55.764: Theme parsing error: gtk.css:128:35: The style property GtkButton:child-displacement-y is deprecated and shouldn't be used anymore. It will be removed in a future version

(solvespace:6659): Gtk-WARNING **: 12:10:55.765: Theme parsing error: gtk.css:129:34: The style property GtkCheckButton:indicator-size is deprecated and shouldn't be used anymore. It will be removed in a future version

(solvespace:6659): Gtk-WARNING **: 12:10:55.765: Theme parsing error: gtk.css:130:36: The style property GtkCheckMenuItem:indicator-size is deprecated and shouldn't be used anymore. It will be removed in a future version

(solvespace:6659): Gtk-WARNING **: 12:10:55.765: Theme parsing error: gtk.css:132:46: The style property GtkScrolledWindow:scrollbars-within-bevel is deprecated and shouldn't be used anymore. It will be removed in a future version

(solvespace:6659): Gtk-WARNING **: 12:10:55.765: Theme parsing error: gtk.css:135:30: The style property GtkExpander:expander-size is deprecated and shouldn't be used anymore. It will be removed in a future version

(solvespace:6659): Gtk-WARNING **: 12:10:55.766: Theme parsing error: gtk.css:142:29: The style property GtkStatusbar:shadow-type is deprecated and shouldn't be used anymore. It will be removed in a future version
SolveSpace!

=================================================================
==6659==ERROR: AddressSanitizer: new-delete-type-mismatch on 0x606000128540 in thread T0:
  object passed to delete has wrong type:
  size of the allocated type:   56 bytes;
  size of the deallocated type: 48 bytes.
    #0 0x7f442094f128 in operator delete(void*, unsigned long) (/usr/lib/x86_64-linux-gnu/libasan.so.5+0xec128)
    #1 0x557e590385f5 in sigc::slot0<bool>::~slot0() /usr/include/sigc++-2.0/sigc++/functors/slot.h:515
    #2 0x557e590385f5 in sigc::slot<bool, sigc::nil, sigc::nil, sigc::nil, sigc::nil, sigc::nil, sigc::nil, sigc::nil>::~slot() /usr/include/sigc++-2.0/sigc++/functors/slot.h:1761
    #3 0x557e590385f5 in SolveSpace::Platform::TimerImplGtk::RunAfter(unsigned int) src/platform/guigtk.cpp:220
    #4 0x557e58f6faa0 in SolveSpace::SolveSpaceUI::ScheduleAutosave() src/solvespace.cpp:312
    #5 0x557e58f91456 in SolveSpace::SolveSpaceUI::Init() src/solvespace.cpp:140
    #6 0x557e59027847 in main src/platform/entrygui.cpp:17
    #7 0x7f441e22309a in __libc_start_main (/lib/x86_64-linux-gnu/libc.so.6+0x2409a)
    #8 0x557e58f6f769 in _start (/usr/local/bin/solvespace+0xab6769)

0x606000128540 is located 0 bytes inside of 56-byte region [0x606000128540,0x606000128578)
allocated by thread T0 here:
    #0 0x7f442094dd30 in operator new(unsigned long) (/usr/lib/x86_64-linux-gnu/libasan.so.5+0xead30)
    #1 0x557e590382c4 in sigc::slot0<bool>::slot0<SolveSpace::Platform::TimerImplGtk::RunAfter(unsigned int)::{lambda()#1}>(SolveSpace::Platform::TimerImplGtk::RunAfter(unsigned int)::{lambda()#1} const&) /usr/include/sigc++-2.0/sigc++/functors/slot.h:546
    #2 0x557e590382c4 in sigc::slot<bool, sigc::nil, sigc::nil, sigc::nil, sigc::nil, sigc::nil, sigc::nil, sigc::nil>::slot<SolveSpace::Platform::TimerImplGtk::RunAfter(unsigned int)::{lambda()#1}>(SolveSpace::Platform::TimerImplGtk::RunAfter(unsigned int)::{lambda()#1} const&) /usr/include/sigc++-2.0/sigc++/functors/slot.h:1774
    #3 0x557e590382c4 in SolveSpace::Platform::TimerImplGtk::RunAfter(unsigned int) src/platform/guigtk.cpp:220

SUMMARY: AddressSanitizer: new-delete-type-mismatch (/usr/lib/x86_64-linux-gnu/libasan.so.5+0xec128) in operator delete(void*, unsigned long)
==6659==HINT: if you don't care about these errors you may set ASAN_OPTIONS=new_delete_type_mismatch=0
==6659==ABORTING
```