solvespace/solvespace #936: Property browser: indicate distance also as vector components

pascal-niklaus, 2021-02-14:
### System information

Solvespace: V3.0

### Expected behavior

This is a feature request:

When a distance is displayed in the property browser (e.g. when two points are selected), it would be nice to not only have the 3d distance but also its projected components along the axis.

I often find myself calculating these from the coordinates displayed, but it seems easy enough to display these also in the property browser.

For example: distance = 5.0 (3.0, 4.0, 0.0)

robnee, 2021-03-01:
@phkahler How does one compute the three components given two point entities?  The UI work should be easy.

phkahler, 2021-03-02:
@robnee how about just ( B.x - A.x ,  B.y-A.y ,  B.z - A.z )

The real pain is dealing with the text window stuff, but this would be for display only so it's a good intro to that.

robnee, 2021-03-02:
@phkahler Thanks!  This works:
```
-        double d = (p1.Minus(p0)).Magnitude();
-        Printf(true, "  d = %Fi%s", SS.MmToString(d).c_str());
+        Vector dv = p1.Minus(p0);
+        Printf(true, "  d = %Fi%s " PT_AS_STR,
+            SS.MmToString(dv.Magnitude()).c_str(), COSTR(dv));
```
I can do a PR tomorrow.  At some point I suppose you are going to freeze feature changes for 3.0 though. :)

phkahler, 2021-03-02:
> I suppose you are going to freeze feature changes for 3.0 though. :)

@robnee  I thought we were feature frozen a  month or two ago. I'm just waiting for the mimalloc stuff and hopefully the Apple M1 builds. It's been way too long since the last "official" release and I want this one to be good for a long time - not that we should wait so long again.

The recent changes are bug fixes and reasonably small  enhancements that have low probability of causing regressions. I don't want to discourage anyone by not merging these things. We shall have our 3.0 final by pi-day, the 14th (I hope).

robnee, 2021-03-02:
@phkahler Ok, thanks.  That's great news about 3.0.  If there's anything I can do to help push it over the finish line please let me know.

phkahler, 2021-03-03:
> If there's anything I can do to help push it over the finish line please let me know.

@robnee  As with all open source contributors, I'm not sure where you came from or why but your contributions are much appreciated! I try not to ask for too much as we're all here by choice. Having said that, if I could pick one issue to fix before 3.0 it would be #937 because I suspect it will be reported more in the future. But the constraints work at least as well as before, so it's not a show stopper. That would also be good experience to have for implementing a new feature I want but haven't even open as issue for yet ;-)

ruevs, 2021-03-03:
@robnee @phkahler just now I started working on #973 - letting you know to avoid effort duplication.

ruevs, 2021-03-03:
I like this a lot, so I made a pull request ;-)

phkahler, 2021-03-04:
@robnee  Well then... I guess back to whatever issues or features you're interested in. We appreciate the contributions!

closed 2021-03-06 by phkahler