solvespace/solvespace #937: Arc of circle tangential to line segment kind of works but is broken

baryluk, 2021-02-15:
### System information

- solvespace from master @ 7795b0bc1911a01eaa4661e252488d8f07ec2135
- Debian testing, amd64
- gcc 10.2.1

[dual-tangent-arc-on-segment.slvs.zip](https://github.com/solvespace/solvespace/files/5980362/dual-tangent-arc-on-segment.slvs.zip)

So apparently, one is supposed to have a line segment that shares the end point with end of an arc, to make tangential constraint work.

But in fact solvespace allows right now to create a tangential constrain without that limtiation. One can for example do a tangential constrain of arc end with any line segment, or with a line segment that has end of an arc on line, but not on end of a segment. It still shows error message, but still creates a constrain and it works.

However, trying to create two such constrains (one for each end of an arc), create a wrong constraints.

https://user-images.githubusercontent.com/110322/107923761-d9210080-6f69-11eb-8ff3-a1294824c1f6.mp4

ruevs, 2021-02-15:
In some sense this is a regression - 2.3 does not do it.

On the other hand being able to apply the tangential in these "more relaxed" cases may be useful. What I think it does is apply the tangential constraint always to the "starting (first) point" of the ark - since if you think about it it is not clear which one should be used with the current UI.

It can potentially be changed to accept an ark, a line segment and one of the end points of the ark. I'm curious when and how this happened, but have no time to dig now... Maybe later.

And the error dialog shuld be rethought in any case.

ruevs, 2021-02-15:
By the way the situation is the same with cubic splines.

ruevs, 2021-02-16:
Aha! I know EXACTLY where this got introduced/broken: 96958f466395148ee7f0a252222a2a27e7345c22 the `return` statements got "swallowed" by the newly created functions. (e.g .https://github.com/solvespace/solvespace/commit/96958f466395148ee7f0a252222a2a27e7345c22#diff-49abc03ed071148c0ebae0c64aafb625fa6223135f77aeecdb47dab4cf8b940cL638 )

Here is the pull request where this was done https://github.com/solvespace/solvespace/pull/833.

The question now is - should  we fix it to work as before (easy) or consider the more "relaxed" functionality? @phkahler @vespakoen ?

Somehow I have a vague memory that allowing what we did by mistake has been requested/discussed in the past (5-6 years ago).

vespakoen, 2021-02-16:
It was definitely NOT my intention to introduce this bug / half feature ;)
I think it would be cool if we can support this use-case, but I am not sure how much work it entails.
I can imagine it could check which "end" to use based on the constraint?

Maybe easiest thing to do is revert the behaviour to the previous one (= throw the error and stop processing) and keep this issue open as a possible future improvement.

ruevs, 2021-02-16:
@jwesthues I'm interested in your opinion on this.

jwesthues, 2021-02-16:
No strong feelings. You can always get that behavior with a construction line (perpendicular to radius for arc, or parallel to line from endpoint to control point for cubic), but slightly nicer to do it without. Selection of the line, curve, and curve endpoint seems like a reasonable UI to me for the latter.

ruevs, 2021-03-06:
Now that the bug is fixed I am reopening this for potentially implementing the feature in the future.

hofmannu, 2024-11-05:
I think this is a really useful feature and would be very nice to have implemented, one use case which I would frequently have is that you make a full circle tangent to a line. The natural workflow for me would be: create a line, create a circle, create a shared point and then add the tangent constraint between the line and the circle.

From my experience with CAD this is a feature that is often used to e.g. position a through hole at equal distance from object edges as shown below.

[image]

ruevs, 2024-11-06:
Implemented in https://github.com/solvespace/solvespace/pull/1493

closed 2024-11-14 by ruevs

PuppyPi, 2024-12-12:
Oh this is it!  This is the feature I've been waiting for that will let SolveSpace do true geometric optics!  :)
(Namely by allowing parabolas (which very particular beziers can be) and their tangent lines (for reflection or normals for refraction) to be included directly instead of painfully manually!)

But, compiling and running the code I suddenly hit an obstacle that seems so obvious maybe it's just me.
..If béziers still don't support the constraint Point-on-Curve, then how can the constraint Tangent-to-Curve (at non-endpoint point) be used in practice?

ruevs, 2024-12-12:
> ..If béziers still don't support the constraint Point-on-Curve,

Indeed - they do not.

> then how can the constraint Tangent-to-Curve (at non-endpoint point) be used in practice?

It can not; because it does not exist. The new functionality is "tangent to curve at the end points without the end points being coincident".

PuppyPi, 2024-12-13:
Ohhhhhh I see, so it's just a convenience for making two construction lines, one at the endpoint of each curve, and setting them parallel.
Okay thanks for the clarification; it's still a nice feature :)

jwesthues, 2024-12-13:
Other point-on-curve constraints are implemented by writing an implicit equation for that curve, and that's hard for general Bezier cubics. It should be feasible for quadratics, and a quadratic entity (perhaps rational) would indeed be useful in many optical applications. We don't have one now, though.

PuppyPi, 2024-12-13:
Oh right I forgot; the béziers don't actually exist in the geometric solver—just the screen and mesh export renderers, so endpoint tangent line constraints are just done by parallelness with the hypothetical line between the endpoint and nearest control point, right?

Thanks for the positive words on it; I'll keep waiting for now :)

PuppyPi, 2024-12-17:
Oh :)
I don't know; I haven't used any that did—I just have done it manually through x/y coordinates and slopes.
But if there was to be a parabolic tool specifically, not just a quadratic bézier tool (or holding down alt and pressing the one for cubic perhaps) which would be constrained into a parabola with construction lines of reflections meeting at a vertex, then there's a few things it would need:
- Non axis alignment; ie, being tilted at any angle
- The capacity for asymmetry (for one side of the focus/vertex to be farther from the focus-vertex axis than another, at any tilt angle)
- Points on it and tangent to it for reflections to be defined as we said before of course.

So given this I think it would still need three points, curve-start, curve-end, but Focus instead of Bézier-control-point would be the most useful for optics.   And I don't know a use of a perfect parabola other than focusing things as in optics (or for a perfect ellipse for that matter), but if there was one, Vertex would be the most useful point for mechanical construction since it's the farthest point of the bend, and the bézier control point would be the most useful for smoothness or tangency at endpoints.

They're all mathematically equivalent, but I don't know if the solver could work with points other than the bézier control point having their own constraints!  Then again you could make that happen with construction lines and/or point-on-curve, so that means the solver could do that automatically and have the three equivalent points of a quadratic element (focus, vertex, bézier control) be present, just constrained to each other (whichever is "the true" quadratic control point), and perhaps a UI could flip between them!

(Personally I'd be fine with any UI though if it made it arrive faster XD )
(Just having parabolas/quadratics and their tangent/normal lines and if possible, ellipses, in any form would make night and day difference!!  I already have to use a bunch of construction lines anyway to manifest reflection or Snell refraction in the solver, which it turns out is actually already possible without an angle-ratio constraint :3 )

commit d97b9bed (2024-11-14): Allow tangent on arcs/lines/splines without coincident endpoints.

Fixes #937