solvespace/solvespace #938: Property browser dissapeared

giannissc, 2021-02-17:
I would like to inform you of a rather weird bug I have found. Sometimes when you work on an external monitor with solvespace the next time you use solvespace without a monitor the property browser is "trapped" into an imaginary monitor. The only way to make it reappper is if you connect it again to a monitor drag both windows to the laptop monitor and disconnect from there on.

I am using Windows 10 (don't know if that helps)

I willing to help any way I can to fix this.

Thanks you for your good work :)

ruevs, 2021-02-17:
Or delete these registry keys:
```
HKEY_CURRENT_USER\Software\SolveSpace\
TextWindow_Bottom
TextWindow_Left
TextWindow_Right
TextWindow_Top
TextWindow_Maximized
```

Or if you are on an old version (2.3) `TextWnd_*`

ruevs, 2021-02-17:
At least on Windows there is code that tries to handle this situation:

https://github.com/solvespace/solvespace/blob/04e872a61fbf1492b5425c0cc674d3a10998496b/src/platform/guiwin.cpp#L1220

So it is interesting how it happened. Can you consistently reproduce it?

ruevs, 2021-02-17:
Fixed.

PR #940 (2021-02-18): Win32: If the main or text window is off-screen make it visible...
...by 1/40th of the scrreen width so that the user can notice it.

This can happen if the window was on a second monitor, which is
disconnected, or if the user knows about `Alt-Space | Move` and then
moving the window with the arrow keys.

Before this, if for example the left edge was off-screen it was moved
to 1920, which is just off-screen, so the window remained invisible.

Fixes: #938
commit 47c3209b (2021-02-18): Win32: If the main or text window is off-screen make it visible...

...by 1/40th of the scrreen width so that the user can notice it.

This can happen if the window was on a second monitor, which is
disconnected, or if the user knows about `Alt-Space | Move` and then
moving the window with the arrow keys.

Before this, if for example the left edge was off-screen it was moved
to 1920, which is just off-screen, so the window remained invisible.

Fixes: #938
closed 2021-02-18 by phkahler