solvespace/solvespace #939: Poor Linking performance

giannissc, 2021-02-17:
I have noticed that the more complex the components become the slower the performance gets in solvespace. This is particularly apparent when linking said complex components. Just changing a size constraint can take up to 10 seconds to complete.

Do you know what might be causing this?
Is there anything I can do to get the most performance out of solvespace?

ruevs, 2021-02-17:
First:
- Use a release build (if you are building from source) - is is much faster than debug.
- Use the OpenMP (multi core) version - uses all the CPU cores for some operations and is somewhat faster for them.

That said SolveSpace does get in some cases. There are three main bottlenecks:
- Too many surfaces and thus triangles to display. You can alleviate this by increasing the `chord tolerance (in percents)` on the configuration screen.
- Too many constraints (complex sketches) - you can not do anything about it. (https://github.com/solvespace/solvespace/issues/841, https://github.com/solvespace/solvespace/issues/225)
- Too many entities (objects) - you can not do anything about it. (https://github.com/solvespace/solvespace/issues/932)

giannissc, 2021-02-17:
Would using a GPU help as opposed to just using the CPU?

phkahler, 2021-02-18:
> Would using a GPU help as opposed to just using the CPU?

Nope. Solvespace has several algorithms that have O(n^2) time complexity. Most of them could be improved to O(n log n) but it is a lot of work and won't really pay of well until we fix all of them.

A highly readable account of a similar situation is here: https://randomascii.wordpress.com/2021/02/16/arranging-invisible-icons-in-quadratic-time/

I've got a really efficient spatial index I'm going to try to dig up that may help some of the issues in solvespace. It was good enough to ray trace things like solvespace models in real time on a CPU so I'm kind of looking forward to locating it ;-)

ruevs, 2021-02-18:
The article is fantastic :-)

phkahler, 2021-02-18:
Linking gets slow I believe because it creates a group with a large number of constraints and only 6 degrees of freedom. Every point in the linked sketch gets a copy created with a rotate/translate transformation applied based on the same 7 parameters. I'm under the impression that the constraint solver/eigen improvements will bring a large performance win in this area.

pjanx, 2021-03-24:
Would money help?

pjanx, 2021-03-27:
I've tried the _eigen_over_master_ branch, and it doesn't help my [cursed model](https://github.com/pjanx/nyquist-case/tree/master/levinson_choc_case), seemingly at all, nor did OpenMP, seemingly at all. There don't seem to be any issues either, though.

gprof on RelWithDebInfo:
```
Each sample counts as 0.01 seconds.
  %   cumulative   self              self     total           
 time   seconds   seconds    calls  ms/call  ms/call  name    
 83.28     39.90    39.90   148330     0.27     0.27  SolveSpace::IdList<SolveSpace::Entity, SolveSpace::hEntity>::Add(SolveSpace::Entity*)
 12.54     45.91     6.01   817479     0.01     0.01  SolveSpace::SolveSpaceUI::CreateDefaultDrawingGroup() [clone .constprop.0]
```

callgrind on RelWithDebInfo:
[image]

callgrind on Debug took roughly a millennium to complete:
[image]

Haven't looked into what it's doing, but it's absolutely dominated by copying entities. Not sure whether [this](https://github.com/Evil-Spirit/solvespace-master/commit/9cf1a09bcdaa5a2bc99400d368dc1a8f43996e1d) would be of any help.

pjanx, 2021-03-27:
So I think I've made my life bearable now. It's much better with this patch:

```patch
diff --git a/src/dsc.h b/src/dsc.h
index 93403bc..76b54c5 100644
--- a/src/dsc.h
+++ b/src/dsc.h
@@ -467,11 +467,16 @@ public:
         // Look to see if we already have something with the same handle value.
         ssassert(FindByIdNoOops(t->h) == nullptr, "Handle isn't unique");
 
-        // Copy-construct at the end of the list.
-        new(&elem[n]) T(*t);
+        int pos = LowerBoundIndex(*t);
+
+        // Shift everything to the right.
+        new(&elem[n]) T();
+        for (int i = n; i > pos; i--)
+            elem[i] = std::move(elem[i - 1]);
+
+        // Copy-construct at the right place.
+        elem[pos] = T(*t);
         ++n;
-        // The item we just added is trivially sorted, so "merge"
-        std::inplace_merge(begin(), end() - 1, end(), Compare());
     }
 
     T *FindById(H h) {
```

[image]

All tests pass and no new leaks seem to be happening.

phkahler, 2021-03-27:
@pjanx  Thank you for looking into this so deep. Your fix is obviously better and will likely be merged very soon. But keep in mind it looks like a constant time speedup, rather than reducing the actual number of list manipulations. That's assuming inplace_merge isn't doing something really stupid. Oh wait, it probably ends up scanning the list looking for the right place to put the new item so if we have a case where that's usually near the end this could be turning O(n) into O(small).

Anyway, thanks again for this efffort!

phkahler, 2021-03-27:
@pjanx  With all the other optimizations that have been done, none of them had much impact on an old sketch of mine. Changing one dimmension took about 1.8 seconds and that's been mostly constant. Your fix bring it to 0.18 seconds!

pjanx, 2021-03-27:
@giannissc can you build from master and verify whether it's bearable in your case now?

giannissc, 2021-03-29:
I can say that it is faster but still not interactive as you do have to wait a few seconds

phkahler, 2021-03-29:
I've added some text on IdList, Entities and the Remap mechanism here:
https://github.com/solvespace/solvespace/blob/master/developer_docs/IdLists_Entities_and_Remap.txt

And I guess I should have manually broken line ;-(

To write that I had to look up some things to make sure I wasn't lying, and it turns out the limit on entities in a group is not due to the global ID list, but the remap structure. Clearly the IdList is a performance issue though, but it should be easier to replace than I thought because it's not the same thing as Remap.

ruevs, 2021-03-30:
Hey I got annoyed and joined the [long and illustrious line of people](https://github.com/solvespace/solvespace/commits/master/src/dsc.h) who have tried to tackle `IdList` :-D

Some highlights ([among many commits](https://github.com/solvespace/solvespace/commits/master/src/dsc.h))

@whitequark 
https://github.com/solvespace/solvespace/commit/c84a3b6de3b83cf5d7d0fc35dd8332fedf37bd9e this one is a regression in performance
@Evil-Spirit
https://github.com/Evil-Spirit/solvespace-master/commit/44eb1a4703a1be50d74fdfeacb97f89c06e08ab6 supposedly fast but never merged and bit-rotting
@rpavlik 
https://github.com/solvespace/solvespace/pull/429 and many others
@phkahler 
https://github.com/solvespace/solvespace/commit/d3da2c8b22d8f84e84adcce329e3b97a6a063e10 an incremental improvement
@pjanx 
https://github.com/solvespace/solvespace/pull/973 an incremental improvement

And finally me - feel free to criticize:
https://github.com/ruevs/solvespace/commits/FixIdList_TheNextAttempt
Still not working. I actually pulled out Strostrup (the physical 1300 page fourth edition) - after getting annoyed with cppreference.com - and it is still not working :-(

Basic idea:
- Elements are stored in the `elemstore` vector **_not_** sorted. And some slots in this vector will be empty/unused after elements are removed. This should avoid expensive copying/moving around/constructing/destructing of large types that are  stored in `IdList`s.
- An `elemptr` vector of pointers to elements sorted by `v.h` provides for the O(log(n)) searching and O[1] `[]` operator.
- A `freelist` vector of pointers to empty elements in `elemstore` avoids having to "pack" it when elements are removed. Empty slots are taken from this list when adding new elements before falling back to `push_back`.

The above (I hope) should give a nice performance boost. 

#### BUT

Unfortunately `NextAfter` becomes slower with this and will probably kill performance even if I make it work. And this stupid function is in 199 places in the code - mostly  for "no good reason" except as a "`i++`" in for loops! The reason it becomes slower is because `IdList` is written to expose its elements as `T *`s and given a `T *` is takes a binary search to find its index in `elemptr` and thus get the next element. To avoid this will take majour refactoring in all users of `IdList` - a typical commit that Whitequesrk would have never merged - and I would have (mostly) agreed with him...

Anyway sorry for the rant - I will play some more when I have some time and am not so sleepy :-)

phkahler, 2021-03-30:
@ruevs you've miss attributed @pjanx improvement to me. I just merged it ;-)

ruevs, 2021-03-30:
> 
> 
> @ruevs you've miss attributed @pjanx improvement to me. I just merged it ;-)

No, I did not, back on Aug 29, 2020 you actually improved `AddAndAssignId` in d3da2c8b22d here is the pull request https://github.com/solvespace/solvespace/pull/687 ;-)

pjanx, 2021-03-30:
@phkahler is just confused here.

phkahler, 2021-03-30:
Some thoughts:

Using NextAfter will be necessary if it gets replaced by anything that doesnt store them consecutively.

Would a red-black tree be a good solution?

It may be a good idea to look at those 199 instances of scanning the whole list. Many of them are searching for items that meet a particular criteria. Unfortunately the list is templated so those searches tend to be specialized to the list they know they're scanning.

Is there a way to profile while ignoring the list operations - roll their time into the caller, so we can see which algorithms one level up are taking all the time?

I know the wishlist includes taking AssemblePolygon from O(n^2) to O(n logn) and it uses these lists if I recall correctly. The list isnt the problem there, but it doesnt help.

ruevs, 2021-03-31:
I think `NextAfter` can be avoided for all the "go through the whole list" for loops
```C++
for(i = whateverIdList.First(); i; i = whateverIdList.NextAfter(i)) {
    i->DoSomething();
}
```
becomes something like:
```C++
for(i = 0; i<whateverIdList.n; ++i) {
    whateverIdList[i]->DoSomething();
}
```
Yes I know this is `C`, but it will(may) be fast.

ruevs, 2021-03-31:
Using a `map` keyed by handle with the value a pointer to the aforementioned `elemstore` may be rather interesting - at least as an experiment. But we lose random access - after I commented the `operator []` of `IdList` I got "only" 13 errors:

```
\src\constrainteq.cpp(219,21): error C2676: binary '[': 'SolveSpace::IdList<SolveSpace::Equation,SolveSpace::hEquation>' does not define this operator or a conversion to a type acceptable to the predefined operator
\src\group.cpp(755,43): error C2676: binary '[': 'SolveSpace::EntityList' does not define this operator or a conversion to a type acceptable to the predefined operator
\src\modify.cpp(412,47): error C2676: binary '[': 'SolveSpace::IdList<SolveSpace::Constraint,SolveSpace::hConstraint>' does not define this operator or a conversion to a type acceptable to the predefined operator
\src\groupmesh.cpp(272,53): error C2676: binary '[': 'SolveSpace::IdList<SolveSpace::SSurface,SolveSpace::hSSurface>' does not define this operator or a conversion to a type acceptable to the predefined operator
\src\srf\merge.cpp(16,25): error C2676: binary '[': 'SolveSpace::IdList<SolveSpace::SSurface,SolveSpace::hSSurface>' does not define this operator or a conversion to a type acceptable to the predefined operator
\src\srf\merge.cpp(34,33): error C2676: binary '[': 'SolveSpace::IdList<SolveSpace::SSurface,SolveSpace::hSSurface>' does not define this operator or a conversion to a type acceptable to the predefined operator
\src\srf\boolean.cpp(206,30): error C2676: binary '[': 'SolveSpace::IdList<SolveSpace::SCurve,SolveSpace::hSCurve>' does not define this operator or a conversion to a type acceptable to the predefined operator
\src\srf\boolean.cpp(698,34): error C2676: binary '[': 'SolveSpace::IdList<SolveSpace::SSurface,SolveSpace::hSSurface>' does not define this operator or a conversion to a type acceptable to the predefined operator
\src\srf\boolean.cpp(704,18): error C2676: binary '[': 'SolveSpace::IdList<SolveSpace::SSurface,SolveSpace::hSSurface>' does not define this operator or a conversion to a type acceptable to the predefined operator
\src\srf\boolean.cpp(712,34): error C2676: binary '[': 'SolveSpace::IdList<SolveSpace::SSurface,SolveSpace::hSSurface>' does not define this operator or a conversion to a type acceptable to the predefined operator
\src\srf\boolean.cpp(853,18): error C2676: binary '[': 'SolveSpace::IdList<SolveSpace::SSurface,SolveSpace::hSSurface>' does not define this operator or a conversion to a type acceptable to the predefined operator
\src\srf\surface.cpp(952,36): error C2676: binary '[': 'SolveSpace::IdList<SolveSpace::SSurface,SolveSpace::hSSurface>' does not define this operator or a conversion to a type acceptable to the predefined operator
\src\srf\surface.cpp(1077,33): error C2676: binary '[': 'SolveSpace::IdList<SolveSpace::SSurface,SolveSpace::hSSurface>' does not define this operator or a conversion to a type acceptable to the predefined operator
```

pjanx, 2021-03-31:
My current idea is maintaining a permutation array next to `IdList` data. That way only sequences of PODs will be moved around, which is fast. Memory complexity will rise (2n/4n bytes) but less than with `std::map`. I don't have the slightest clue about consequences around the codebase, it's just something I want to try and shouldn't be difficult.

pjanx, 2021-03-31:
Anything non-obvious is definitely post-3.0, the improvement over the last stable version is overall huge, and I personally hope to see it out of the door soon.

pjanx, 2021-04-03:
@giannissc see #992 if you can build from sources, it should go a step further.

giannissc, 2021-04-07:
Most of my performance issues from the past models have been solved using another pull request (i can't remember now which one. I think it some something about slow translate groups). I am currently working on a much bigger and compelx model that should serve as good reference point for performance. Will keep you posted :)

ruevs, 2021-04-07:
@giannissc A (potentially) even faster version is coming. But it will be after 3.0 is released. For now you can try it by compiling from source https://github.com/ruevs/solvespace/tree/OptimizeIdList 
https://github.com/solvespace/solvespace/pull/1002
And another one that does a pretty similar thing:
https://github.com/solvespace/solvespace/pull/992

phkahler, 2021-05-13:
With one of the PRs merged can we close this now? Try the latest edge build when it lands.

ruevs, 2022-01-06:
@phkahler I think we can close this after https://github.com/solvespace/solvespace/pull/1002, https://github.com/solvespace/solvespace/pull/1158, https://github.com/solvespace/solvespace/pull/1159 are merged and https://github.com/solvespace/solvespace/issues/386 is closed.

@giannissc would you please try the cutting edge build from here https://github.com/solvespace/solvespace#via-automated-edge-builds with one of your "slow" models and give us feedback? If it is still slow it would be great if you can attach and example.

phkahler, 2022-01-23:
Another thing that might help is to "hide edges of solid model". It's the 8th button at the top of the text window.

ruevs, 2022-06-04:
@giannissc @pjanx @rpavlik @phkahler I am closing this. If you have some big models that make SolveSpace (3.1+) unreasonably slow - please open a new issue.

closed 2022-06-04 by ruevs