solvespace/solvespace #943: Clarify license

jtojnar, 2021-02-25:
The README says SolveSpace is licensed under GPL 3, which is unclear; see https://www.gnu.org/licenses/identify-licenses-clearly.html
Could you clarify whether you mean `GPL-3.0-only` or `GPL-3.0-or-later`. The source code seems to say the latter:

https://github.com/solvespace/solvespace/blob/41c81a71c1c432c2d5ff3342d2a570c19d01727b/src/solvespace.cpp#L1002-L1004

ruevs, 2021-02-25:
@jwesthues ?

jwesthues, 2021-02-26:
Thanks, good catch. I've updated the ambiguous references in README.md and CONTRIBUTING.md to align with the unambiguous about box text, in 2fcc933aa9c7d8f30a35d94fdb760b343e4562bf. I notice that Debian has us as "GPL-3.0+", so that's also consistent.

I believe it's very unlikely that any past contributor both (a) missed the "or later" in the about screen and thought based on the ambiguous references that we were "only", and (b) cares about the difference. If anyone does have a concern then please let us know, though.

closed 2021-02-27 by ruevs