solvespace/solvespace #947: MinGW build broken after mimalloc 2.0.0 update

nirvdrum, 2021-03-02:
### System information

- **SolveSpace version:** 
- **Operating system:** 

Ubuntu 20.04 x86_64 under WSL2.

### Expected behavior

SolveSpace should build the Windows 64-bit binary. I've been following the process in the README and have been successfully building and using SolveSpace for several months.

### Actual behavior

After the mimalloc 2.0.0 update, SolveSpace ceased building on Linux. The issue was [reported](https://github.com/microsoft/mimalloc/pull/367) and [fixed](https://github.com/microsoft/mimalloc/commit/71ac98ab089fc5380335486eb449578fd6cf20c0). upstream. I don't know what mimalloc's release plan is, but SolveSpace won't cross-compile with MinGW until that patch is applied.

### Additional information

If it's of interest, I could add steps to the GitHub Actions configuration to cross-compile SolveSpace on Ubuntu. Obviously it would fail currently, but it could help reduce the likelihood of future regressions.

phkahler, 2021-03-02:
@nirvdrum  There are two other fixes in the mimalloc dev_slice branch that we want, one for Apple M1 and one for the -march=native issue. They are going to push out a new release any day now (he said "this week" but that was last week) and I hope to bump our mimalloc version to whatever that is. It should fix our remaining build issues and then we'll have our 3.0 final soon after.

Would the GitHub action for cross-compiling be to find issues like this one earlier?

nirvdrum, 2021-03-02:
Thanks for the update. I was hoping we could get the MinGW update in before the SolveSpace 3.0.0 release.

As for the GitHub Action, I think it would have caught the issue with the mimalloc update at the time of the update. Of course, if cross-compilation isn't considered a primary use case, you may not want it blocking updates like that.

phkahler, 2021-04-03:
@nirvdrum  I just updated our mimalloc version. Can you try a MinGW build again? Reopen this if there is still a problem.

closed 2021-04-03 by phkahler

nirvdrum, 2021-04-03:
@phkahler I was able to build cleanly with MinGW an Ubuntu 20.04 WSL2 instance. Thanks.

PR #989 (2021-04-03): Update mimalloc version.
This close #947   close #934

Help with #913   - no patches required.