solvespace/solvespace #950: Bug: boolean intersection operator for solid model not available

matt-hammond-001, 2021-03-04:
### System information

- 3.0~72635df5
- Mac OS X 10.15.7

### Expected behavior

Create a solid - e.g. a simple square extruded to a cube.
When I have the extrusion selected, I expect to see "intersection" as an available boolean operator

### Actual behavior

Boolean "intersection" operation is not shown as one of the options to select. Only "union", "difference" and "assembly" are shown.

### Additional information

I understood from #35 that boolean intersection has been implemented and merged into master. I therefore do not know why it is unavailable to me.

jelle-r, 2021-03-05:
I am having the same problem in version 3.2 on ubuntu. I see only "union" and "difference" as options for my extrude.

ruevs, 2021-03-05:
Your version is too old - 72635df5 is from 2020-04-02. Intersection was introduced in 3888909d02 on 2020-05-08.
Use a fresh build from Edge here https://github.com/solvespace/solvespace/releases

closed 2021-03-05 by ruevs