solvespace/solvespace #951: Tangency between 2 Bezier splines

Tanzit555, 2021-03-07:
Hi guys,

  When I try to set a tangency between 2 Bezier splines connected together by their extremities, It only sets it on one connection, not the other... Normal behavior ???

Thanks for your replies

[image]

### System information

- **SolveSpace version:** 
- **Operating system:**

robnee, 2021-03-07:
I'm pretty hopeful that this can be fixed.  It's really two separate constraints, one for each pair of endpoints.  The problem is when attempting to apply the constraint the second time SolveSpace sees it as redundant and ignores the request.  The PR demonstrates that this can be fixed but still needs some work.

phkahler, 2021-03-07:
The work around is to make a construction line from the end of a curve to its adjacent control point. Then constrain the closest control point from the other curve to that construction line.

@robnee The fix for this may have to wait until after 3.0 because we don't want to break any existing sketches if it later needs to be tweaked. But we'll see ;-)

robnee, 2021-03-07:
@phkahler Yes, this is trickier that it first appeared.

ruevs, 2021-03-07:
Related to #937.

Evil-Spirit, 2021-04-22:
Don't forget about 2nd order tangent

ruevs, 2024-11-06:
With #937 is implemented in https://github.com/solvespace/solvespace/pull/1493 is is now possible to do this by first making the two ends tangent and then making them coincident. I should still look into why it does not work normally.

ruevs, 2025-05-17:
After the [above fix](https://github.com/solvespace/solvespace/issues/951#issuecomment-2460221949) it is possible to do this (constrain both ends tangent). To which point the constraint is applied depends in the order in which the beziers are selected. And I think this makes sense.

In my opinion we should close this @phkahler what do you think?