solvespace/solvespace #955: Hole Tool - feature

phkahler, 2021-03-07:
## Hole Tool: Countersunk, Counterbore or Simple holes

## Proposed operation (3D groups):
A new sketch tool would be added. Once selected and the mouse hovers over a plane face, that face is highlighted as usual. When left mouse click, a hole entity is created in the 3D sketch group AND the solid model is modified. Multiple holes could be created in a single group.

The sketch entity might be a circle with cross-hairs and a center point. The point would be constrained to the face clicked on and the normal for that face would be used to orient the entity and hole. The point could then be dragged or constrained as usual.

## implementation
One key issue is how to define a hole. Using 2 diameters, a depth (of the counterbore), and a checkbox (counterbore, countersink, normal) would cover the basics. Some people prefer to specify a countersink by the angle as on the [wikipedia page](https://en.wikipedia.org/wiki/Countersink). I assume the definition will be in the text window. It's not clear to me how to define the final depth - should it be parametric (per hole) like a normal extruded hole, or defined in the text window? What about non-through holes?

If the depth is draggable/constrainable, an additional line or circle may be needed but will these will all need to be part of the single hole entity. Also, how would that work with the 2D option below?

Once a hole has been created, its definition should be used for subsequent holes until one is edited. It might also be possible to copy/paste a hole entity, which would bring along its parameters but take its normal from the new location.

## 2D option
People will probably want to use the new sketch entity for 2D drawings. and of course they will want those holes to be formed if they extrude the sketch. The 3D option is more universal but not as nice for simple designs mostly covered by a 2D drawing, so this seems important. Biggest issue is that the 3D NURBS behavior would be quite different between the two. Hole would be created from previous 2D sketch entities AND current 3D group entites, both after the solid model is created. Maybe that's not so bad.

### Hole styles?
Would it be feasible to allow the user to create named hole definitions, similar to line styles? Would the hole style be shown on the sketch?

### Key questions:
1) what is the best way to define a hole profile?
2) how to handle the final depth of the hole?
3) are there any other important types of holes to handle?
4) is this too much violation of the SolveSpace group structure?

ruevs, 2021-03-07:
This is a "weird" idea in context of the "ideology" of SolveSpace so far. Making "holes" with their "machining" features is a common and useful task, but so far all tools have been pure geometry. If we start down this path it leads to treads, bolts, nuts, axles, gears...

jwesthues, 2021-03-08:
I like the feature, but it feels more like a group than an entity to me? That keeps the idea that entities never directly affect the solid model, and would permit the group to generate e.g. standard circle and point entities to be constrained against later in the model. I think that sets a reasonable precedent for the future too--a gear entity seems bad to me, but a gear group seems perfectly fine.

As to depth, I suspect that either fixed or "all the way through" covers most use. You could implement the latter by adding an extrude option that extrudes to the next surface and stops, which the pro CAD tools all have. Ultimately it would be nice if the depth etc. could be specified parametrically, perhaps as part of the broader symbolic expression idea that I intended but never implemented, for example `length(e1234) + 5` for the length of entity e1234 plus 5 of the current units.

phkahler, 2021-03-11:
I have a real problem with making a new group for each hole. Also, editing parameters of an entity (request?) in the text window would be similar to editing group properties. After asking an actual professional CAD user about defining holes with features and consulting google with the right words, I found standard annotation of holes.

Counterbore:
&#x2300;10 &#x21A7;25.0
&#x2334;&#x2300;20 &#x21A7;10.0

Countersink:
&#x2300;10 &#x21A7;25.0
&#x2335;&#x2300;20x120.0&#x00B0;

Yes, there are unicode symbols for counterbore (2334), countersink(2335), and depth(21A7). Apparently omitting the depth value implies a through hole. Not sure if SolveSpace will produce these symbols yet.

I would say it's appropriate to use these in a 2D drawing and it would be great to have the holes appear when extruded - I can make that happen ;-)  It still seems like a good idea to have holes available in a 3D group as well, but there might be some differences.

@jwesthues Do you think we could create several 3D circles from a single hole entity, similar to how a point becomes a line when extruded, or a line becomes a face? The top and bottom circles will have different diameters, and there may be one or two inside a hole as well.

jwesthues, 2021-03-11:
I think most commercial CAD tools have a hole tool that appears in their feature tree in the same way as an extrude or cut would? That would make them more analogous to SolveSpace groups than to entities. I'm imagining that a single group could be multiple holes, just all with the same diameter, depth, etc.

It would certainly also be possible for the extrude to recognize hole entities and special-case them. I don't hate that, but it's inconsistent both with the behavior of existing SolveSpace entities (which form a strict solid of extrusion when extruded) and with holes in commercial CAD.

phkahler, 2021-03-13:
I think I'm trying to cover 2 different use cases. One is to allow proper hole definitions in technical drawings - issue #961 will need to be handled first for that. But once we have a way to define a hole, there should be a way to create them on the 3D model. The FreeCAD tool uses a hole group but takes all circles in a 2D sketch to makes holes - ignoring the diameter of the circle and imposing the hole parameters defined for the group. Description here:
https://wiki.freecadweb.org/PartDesign_Hole

Their hole definitions are more complete than I was thinking (or even aware of).

I think it's OK if holes are handled differently, but I think they should be done well. We should be able to do better than what FreeCAD is doing with circles.

It looks like some might even be parsing the annotation info to create really complex holes:
https://www.innova-systems.co.uk/wp-content/uploads/SOLIDWORKS-2018-Advanced-Hole-Callout-15.png

I'm still very much open to suggestions on this ;-)

phkahler, 2023-01-25:
Started holes branch on my fork:
https://github.com/phkahler/solvespace/tree/holes

It uses "z" to create a hole request/entity. These need to be construction only (or a hole style?)
Next todo:
 * [ ] Figure out how to store the hole profile/parameters
 * [ ] Add an annotation. These should be sharable among holes, but how?
 * [ ] Match the parameters/annotation/visual
 * [ ] Create the hole geometry when extruded

Not sure if we should render stringers from each hole to the annotation?

phkahler, 2023-02-05:
@jwesthues  What is the difference/relationship between a request and an entity?  I want a "hole" entity to have new data:   diameter1, diameter2, depth, depthAngle, and featureType (emun). These will not be constrainable on the sketch, but will be edited via the text window. They need to copt/paste and save to files. Where is the best place to put them? I'm thinking in the Entity or EntityBase but not sure.

jwesthues, 2023-02-05:
When the user chooses Sketch -> something, that creates a request. When the group containing that request is generated, that request generates one or more entities. For example, a request for a line segment generates entities for its two endpoints, and then an entity for the line connecting them. A request for a circle generates an entity for its center point plus some other stuff. Since the same point entities are used for both, the same constraints can work on both. Entities can also be generated in other ways, for example when a point in a sketch gets extruded into a line, or from a linked part in an assembly, or from a step and repeat. (I chose the name "request" because the user is directly requesting that we generate specific entities, vs. entities generated indirectly later).

So you could pretty naturally define a new hole request. For a simple hole, that request might generate circles at the top and bottom, points defining the centers of those circles, distances defining the diameters of those circles, normals defining the planes of those circles, and a line between the points. For a complicated counterbore or countersink, that request might generate yet more entities; but since they're all standard existing entities, the standard existing constraints will all work as expected without further effort.

The interaction with the solid model is trickier, since neither requests nor entities do that directly now, only groups like lathe or extrude or helix. You could define a new hole group to apply that change, though making that work correctly with stuff like step and repeat may require some thought. One option would be to let the hole request create a consistent pattern of entities, and then infer the desired solid hole from the entities (not the request, since it would be painful to follow back to that through a step and repeat, link, or other derived geometry)--so the two coaxial circles become a simple hole, three become a countersink, etc.

phkahler, 2023-02-06:
@jwesthues  I'm trying to decide if I need to add some data members to Entity class, for the 2 diameters, and depth/angle for a featured hole. I want 2D sketches to have a complex hole that behaves like a single entity, not a collection of them contained in a Request (at least I don't think I want that). The idea in 2D is to support technical drawings of essentially 2D parts with such holes, and later get the G-code export to support the counter-bore and counter-sink complete with automatic tool changes. All of that from a single 2D sketch (not sure where we'll include material thickness).  I also want to be able to copy/paste a hole in a 2D sketch, and have the annotation shown or not.

jwesthues, 2023-02-07:
As far as most users are concerned, I'd guess e.g. that a circle now behaves like a single "entity", even though it's actually four `Entity`s behind the scenes. So I agree that the hole should behave like a single thing, but I don't think that requires it to be implemented as a single `Entity`. The basic unit of editing for the user is a `Request`, and the `Entity` is an implementation detail they don't usually care about. (That's somewhat confusing given the use of "entity" in other CAD programs. In retrospect, perhaps the `Request` should be called a `UserFacingEntity`, and the `Entity` should be called an `InternalGeometryPrimitive`.)

You could also create a new `Entity` for the hole, but that would still most naturally be defined in terms of existing `POINT_xxx`, `DISTANCE_xxx` and `NORMAL_xxx` entities, in the same way that a circle is now. If it weren't, then none of the existing constraints would work on it, nor would any of the existing transforms (step and repeat, link, etc.)--so all that work would need to be duplicated. The advantage of the new `Entity` would be that your solid geometry code could work from that directly, instead of inferring from the generated pattern of circles and lines.

jwesthues, 2023-02-08:
Yeah, you'd need to extend that table and/or special-case the new request. That should be easy, though.

The map from `hRequest` to initial `hEntity` and `hParam` doesn't use a remapping table. We need that table to map `hEntity` to derived `hEntity` in groups like step and repeat or link, since those can be nested arbitrarily deep; but since the map from `hRequest` is only ever one step, it's just a crude bitfield.

phkahler, 2023-02-16:
Just an update on my thinking. I was previously thinking a 2d sketch with a hole request would be a sort of special case on extrusion. Yesterday I realized the thing to do is simply create a 3D hole request in the new group upon extrusion and let the 3D version of it create the hole ;-) However, it doesn't look like we create requests when extruding. A lot of entites are copied to the top and bottom, and a lot of lines are created connecting them, but no requests. So this "copy a 2d hole request to the 3d group" would be something newish. @ruevs what do you think about all this? Or are you still not sure about it?

jwesthues, 2023-02-17:
> Yesterday I realized the thing to do is simply create a 3D hole request in the new group upon extrusion and let the 3D version of it create the hole ;-) However, it doesn't look like we create requests when extruding.

The point of `Request`s is that they're always user-requested. The `Requests` are what the user has drawn explicitly, and they're stored when we save the file. Then the `Request`s generate `Entity`s when we `Generate()`, and `Entity`s also generate other `Entity`s (like when a point gets extruded to a line). That `Entity` structure doesn't need to be stored, since we can always regenerate it. (It does get stored for the benefit of assemblies, but that's not used when we're editing the part itself.) Nothing ever creates a `Request` except the user.

The behavior that you want might map naturally to a new hole `Request` and `Entity`--the user creates the hole `Request`, which generates the hole `Entity` in the sketch group, which then generates additional `Entity`s when extruded.

DjogaRo, 2024-04-27:
Sounds more like content for a library of generic pre-defined parts instead of a core concept?
If so, the question would be what's needed beyond hierarchical, parametric sketches.
Parts would need to (be able to) carry some metainformation like can it be linked in 2d, 3d; if so, freely or what against - like for those holes, in 2d it needs to be within a contour and its top face would automatically be contraint to coincide with the workplane, in 3d it needs to be placed on a solids surface and be oriented inwards.
Then how would it be represented - in 2d with symbols and annotation, in 3d as a subtractive solid.

Then such a library could contain as a parametric part a counterbore hole with rather free parameters. Another part in the library could be parameter wrapper for the first one defining certain standard diameters as valid values. Maybe two of those wrappers, one for metric standard stuff, one for the U.S. There could be parts being combinations of bores and threads. And so on.

Feasible? Implementing the basics for such a concept, I mean.

ruevs, 2024-04-28:
Personally I also think that a dedicated "hole tool" (and other similar tools for creating "engineering" features that are not pure geometry) goes against the minimalistic philosophy of SolveSpace.

Your ice - for holes to be linkable from a standard library of "components" of "features" is good. Currently the geometry of linked models can not be modified from the file that links them. So while it may be possible (albeit impractical) to have a huge number of models for every possible screw, it is completely impractical for holes. So if/when one day someone writes the code to make it possible to parametrise linked models your idea will certainly be a better approach then a dedicated "hole tool". At least in my opinion.

DjogaRo, 2024-04-28:
I would have thought the hierarchical, parametric sketches that are in the works would go in that direction?

ruevs, 2024-04-28:
They are not "in the works" - at least the "huarchical" part is not. "Parametric" as you've noticed has significant progress lately.

Contributions are always welcome.