solvespace/solvespace #957: stipple width

Tanzit555, 2021-03-07:
Hi guys,

In the settings, "stipple width" is the area where the targeted element is selected right ?
When I modify it in the settings, it doesn't keep the setting saved, when I restart SolveSpace, it returns back to the default value (15).
All the other settings I modify are kept.

Any ideas ??

Thanks a lot...

Win 10 home 64

3.0RC2

ruevs, 2021-03-07:
The `stipple width` and `stipple type` are currently only saved for custom styles, which are stored in the .slvs files.
For the default styles, which are stored in the configuration - they are not remembered. I do not know if this is intentional.

phkahler, 2021-03-08:
Some key questions here.

Should a .slvs file look the same on every installation, or should the default styles be a user preference like colors in a text editor? The answer to this tells us if modified defaults should be saved in the .slvs file or not. Another question: should the defaults be editable at all? I think the answer is probably yes.

This is a good time to bring this up with @robnee doing PR #952 which is related to when styles do what. Although that is in regard to 2D drawing exports.

robnee, 2021-03-08:
I think it's reasonable to expect the stipple settings to be saved with the other default style settings like color, line width and text height.  @ruevs has a point, it's a little strange that they are not already saved although I can't think of a use case where this would be desired.  Many of the default styles for things like HOVERED, SELECTED, ERROR_COLOR seem more like user settings than model settings.  I don't think saving those to the .slvs file makes sense.  Users already have freedom to style entities with custom styles if they want those settings saved.

Symbian9, 2021-03-09:
> Should a `.slvs` file look the same on every installation, or should the default styles be a user preference like colors in a text editor?

Yes. As for me, it is related to feature request described in https://github.com/solvespace/solvespace/issues/902 and there should be next options for style presets in *"Property Browser > line styles"*:

- Reset to factory defaults (`load factory defaults`, but with more clear naming)
- Save line styles to file...
- Load line styles from file...

So, user would be able to switch between styles presets in much easier way, without editing each line style every time it want to set *custom rendering style* or render same model with various *custom rendering style*.

ruevs, 2021-03-17:
Sure go-ahead. I see no reason why they are not saved.

robnee, 2021-03-25:
@phkahler @ruevs I've made the stipple styles persistent, it's fairly simple.  For example the json config files on Linux have new entries like this:
```
  "Style_ActiveGrp_StippleType":"Continuous",
  "Style_ActiveGrp_StippleScale":30.0,
```
I'm thinking that should be it's own PR separate from save/load:

> * Save line styles to file...
> * Load line styles from file...

I have a couple of questions about that implementation since some of this has no precedent elsewhere in SS.  I was planning on saving these as a json files with 6 elements per style in the same schema as the Linux config.
```
  "Style_ActiveGrp_Color":16777215,
  "Style_ActiveGrp_Width":1.5,
  "Style_ActiveGrp_StippleType":"Continuous",
  "Style_ActiveGrp_StippleScale":30.0,
  "Style_ActiveGrp_TextHeight":11.5,
  "Style_ActiveGrp_Exportable":true,
```
This would require a json library for all platforms for GUI builds.  Currently only the Linux build has a dependency on json.   This might result in the need for a new platform abstraction so the save/load would not depend on any one json API. I think json-c is available for Windows but not sure about MacOS.  Can you think of an alternate format with fewer kinks?

Should the import and export of styles be options offered in "Property Browser > line styles" or on the File menu?  Should custom styles be included?

I think save/load for styles is a worthwhile feature but probably not for 3.0 if it requires build changes.

Symbian9, 2021-03-25:
> This would require a json library for all platforms for GUI builds.

Why JSON, is not such string as `"Style_ActiveGrp_Color":16777215,` is a pure CSV?

ruevs, 2021-03-26:
> I'm thinking that should be it's own PR separate from save/load:

@robnee definitely a separate pull request.

And the saving/loading (I would call it exporting/importing) of the "per user local options" (registry on Windows/ json elsewhere) is such a niche feature that I would definitely not worry about it.
Certainly not for 3.0.

PR #974 (2021-03-29): Save stipple settings for default styles to config
This PR saves the stipple settings for default styles to the configuration and closes #957.
closed 2021-03-29 by phkahler