solvespace/solvespace #960: Segmentation fault on opening SLVS file saved in previous session

Symbian9, 2021-03-09:
### System information

- **SolveSpace version:** 3.0~ https://github.com/solvespace/solvespace/commit/33091817738d780d904bcd3e077022d87ebf2cd3
- **Operating system:** Debian 10 buster

### Expected behavior

`.slvs` file saved with SolveSpace should be opened in same version of SolveSpace.

### Actual behavior

SolveSpace crashed just on opening `.slvs`, which was created just with it.

```Shell
$ solvespace
SolveSpace!

(solvespace:13293): Gtk-WARNING **: 19:09:33.924: gtk_window_present_with_time() should not be called with 0, or GDK_CURRENT_TIME as a timestamp, the timestamp should instead be gathered at the time the user initiated the request for the window to be shown
Generate::REGEN (for bounding box) took 96 ms
Generate::REGEN took 193 ms
Generate::ALL (for bounding box) took 108 ms
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
Generate::ALL took 360 ms
Generate::DIRTY (for bounding box) took 80 ms
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
Generate::DIRTY took 199 ms
Segmentation fault
```

### Additional information

- SLVS (direct link; use "Save As..." to download) - https://github.com/Symbian9/SolveSpace-Daily-Engineering/raw/master/DailySolveSpace/dailysolvespace-20210309_1.slvs

- Screenshot taken from previous session when file created & than saved:
[image]

robnee, 2021-03-10:
I could not reproduce this on linux using this model.  I tried running through valgrind as well and didn't turn up anything interesting.  Could be related to #935.

Symbian9, 2021-03-10:
> I could not reproduce this on linux using this model.

Is it opens for You?

@robnee, Could you try to open it with a Debian/Ubuntu DEB package from @AlexanderP's PPA:

- https://notesalexp.org/
  - https://notesalexp.org/packages/en/buster/amd64/solvespace/

phkahler, 2021-03-11:
The file opens fine for me on Fedora 33 Linux.  It is probably related to the other Debian crash, but I don't know what could be causing it. The Mimalloc stuff was for ARM, not intel, but they will have an update hopefully soon with several fixes we want.

Symbian9, 2021-03-12:
>  I don't know what could be causing it.

@phkahler, FTR, SolveSpace failed only on this testcase `.slvs` file actually. Here is backtrace:

<details>
    <summary>GDB Backtrace Log</summary>
$ gdb solvespace
GNU gdb (Debian 8.2.1-2+b3) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from solvespace...(no debugging symbols found)...done.
(gdb) run
Starting program: /usr/bin/solvespace 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffedae7700 (LWP 9689)]
[New Thread 0x7fffed2e6700 (LWP 9690)]
SolveSpace!

[New Thread 0x7fffe634b700 (LWP 9691)]
[New Thread 0x7fffe5a09700 (LWP 9692)]
[New Thread 0x7fffe5208700 (LWP 9693)]
[New Thread 0x7fffe4a07700 (LWP 9694)]
[New Thread 0x7fffdb816700 (LWP 9695)]

(solvespace:9682): Gtk-WARNING **: 16:23:44.965: gtk_window_present_with_time() should not be called with 0, or GDK_CURRENT_TIME as a timestamp, the timestamp should instead be gathered at the time the user initiated the request for the window to be shown
[New Thread 0x7fffd890b700 (LWP 9697)]
[New Thread 0x7fffc260f700 (LWP 9698)]
[New Thread 0x7fffc1e0e700 (LWP 9699)]
[New Thread 0x7fffc160d700 (LWP 9700)]
[New Thread 0x7fffc0e0c700 (LWP 9704)]
[New Thread 0x7fffabfff700 (LWP 9705)]
[New Thread 0x7fffab7fe700 (LWP 9706)]
[New Thread 0x7fffaaffd700 (LWP 9707)]
[New Thread 0x7fffaa7fc700 (LWP 9708)]
[New Thread 0x7fffa9ffb700 (LWP 9709)]
[New Thread 0x7fffa97fa700 (LWP 9710)]
[Thread 0x7fffa9ffb700 (LWP 9709) exited]
[Thread 0x7fffaaffd700 (LWP 9707) exited]
[Thread 0x7fffc260f700 (LWP 9698) exited]
[Thread 0x7fffaa7fc700 (LWP 9708) exited]
[Thread 0x7fffa97fa700 (LWP 9710) exited]
[Thread 0x7fffab7fe700 (LWP 9706) exited]
[Thread 0x7fffc1e0e700 (LWP 9699) exited]
[Thread 0x7fffd890b700 (LWP 9697) exited]
[Thread 0x7fffc0e0c700 (LWP 9704) exited]
Generate::REGEN (for bounding box) took 136 ms
Generate::REGEN took 224 ms
Generate::ALL (for bounding box) took 108 ms
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
Generate::ALL took 379 ms
Generate::DIRTY (for bounding box) took 101 ms
bad, edge_inters=4
[... 67 lines cut ...]
#56 0x00007ffff7284941 in gtk_main_do_event () from /lib/x86_64-linux-gnu/libgtk-3.so.0
#57 0x00007ffff6f84455 in ?? () from /lib/x86_64-linux-gnu/libgdk-3.so.0
#58 0x00007ffff6f9499e in ?? () from /lib/x86_64-linux-gnu/libgdk-3.so.0
#59 0x00007ffff6f95b16 in ?? () from /lib/x86_64-linux-gnu/libgdk-3.so.0
#60 0x00007ffff6f95cd4 in ?? () from /lib/x86_64-linux-gnu/libgdk-3.so.0
#61 0x00007ffff6e42c8d in g_closure_invoke () from /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#62 0x00007ffff6e56365 in ?? () from /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#63 0x00007ffff6e5f2be in g_signal_emit_valist () from /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#64 0x00007ffff6e5f97f in g_signal_emit () from /lib/x86_64-linux-gnu/libgobject-2.0.so.0
#65 0x00007ffff6f8db6d in ?? () from /lib/x86_64-linux-gnu/libgdk-3.so.0
#66 0x00007ffff6f78c08 in ?? () from /lib/x86_64-linux-gnu/libgdk-3.so.0
#67 0x00007ffff6d61863 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#68 0x00007ffff6d60dd8 in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#69 0x00007ffff6d611c8 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#70 0x00007ffff6d614c2 in g_main_loop_run () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#71 0x00007ffff7283a15 in gtk_main () from /lib/x86_64-linux-gnu/libgtk-3.so.0
#72 0x0000555555597baa in ?? ()
#73 0x00007ffff64bf09b in __libc_start_main () from /lib/x86_64-linux-gnu/libc.so.6
#74 0x000055555559cc0a in ?? ()
(gdb) 
```

</details>

Also tested to create other `.slvs` file with just a *simple cube* — and SolveSpace reopens it successfully.

rpavlik, 2021-03-24:
Weird, I don't even see SolveSpace in the backtrace, presumably it's some of the mystery entries and was build without symbols? Can you load the file to eg export to image, with SolveSpace cli? That would be a bit easier to step thru in a debugger if so.

Symbian9, 2021-03-24:
> was build without symbols?

Yep, @AlexanderP's PPA provide only DEB builds *without debug symbols*.

> Can you load the file to eg export to image, with SolveSpace cli?

@rpavlik, What exact `$ solvespace-cli <...>` command should I use for this?

rpavlik, 2021-03-25:
Had to check, I responded from mobile last time. Any of the commands should work, really. I just used the built-in help. Something like this will work as long as there's supposed to be a solid in there:

```sh
solvespace-cli export-mesh --output %.stl yourfile.slvs
```

Symbian9, 2021-03-25:
> Something like this will work as long as there's supposed to be a solid in there:

I just updated to *SolveSpace* `3.0~` https://github.com/solvespace/solvespace/commit/10ca307d986ab0c52faaab4fcaedb26bd42d1fd6 and here are results (TL;DR: it still *reproducible* **IF** try *open this SLVS via file chooser dialog*)

### Successful

- `$ solvespace *.slvs` can read read this SLVS & open it:  **Success**
<details>
    <summary>Terminal log ($ solvespace ./*.slvs)</summary>

```
$ solvespace ./dealwithit-glasess.slvs
SolveSpace!

Generate::ALL took 31 ms

(solvespace:30331): Gtk-WARNING **: 17:00:54.282: gtk_window_present_with_time() should not be called with 0, or GDK_CURRENT_TIME as a timestamp, the timestamp should instead be gathered at the time the user initiated the request for the window to be shown
Generate::REGEN (for bounding box) took 201 ms
Generate::REGEN took 435 ms
Generate::ALL (for bounding box) took 198 ms
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
Generate::ALL took 487 ms
Generate::DIRTY (for bounding box) took 108 ms
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
Generate::DIRTY took 253 ms
```

</details>

- `solvespace-cli <command> <options> <filename> *.slvs` can read this SLVS & export it STL: **Success**

<details>
    <summary>Terminal log ($ solvespace-cli export-mesh --output name.stl file.slvs)</summary>

```Shell
$ solvespace-cli export-mesh --output ./test.stl ./dealwithit-glasess.slvs
Missing (absent) translation for group-name'#references'
Missing (absent) translation for group-name'sketch-in-plane'
Generate::REGEN (for bounding box) took 132 ms
Generate::REGEN took 231 ms
Generate::ALL (for bounding box) took 137 ms
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
Generate::ALL took 363 ms
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
Generate::ALL took 194 ms
Written './test.stl'.
```

</details>

### Failed

BUT if SolveSpace still failed if try to open this SLVS via file chooser dialog

1. Launch SolveSpace in GUI mode (via Terminal or via Launch icon)
2. Open *"File > Open..."* dialog;
3. Select this SLVS file (i.e. `dealwithit-glasess.slvs`) and click <kbd></kbd> button;
4. SolveSpace crashed (in terminal it shows `Segmentation fault`)

```Shell
$ solvespace
SolveSpace!

(solvespace:30405): Gtk-WARNING **: 17:04:28.931: gtk_window_present_with_time() should not be called with 0, or GDK_CURRENT_TIME as a timestamp, the timestamp should instead be gathered at the time the user initiated the request for the window to be shown
Generate::REGEN (for bounding box) took 109 ms
Generate::REGEN took 227 ms
Generate::ALL (for bounding box) took 125 ms
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
Generate::ALL took 430 ms
Generate::DIRTY (for bounding box) took 83 ms
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
Generate::DIRTY took 215 ms
Segmentation fault
```

rpavlik, 2021-03-25:
oh that's very interesting, so it is actually related to the file dialog somehow.

robnee, 2021-03-25:
Do we know if gtkmm is > 3..24.0?  i.e. using the native file chooser?

Symbian9, 2021-03-25:
> Do we know if gtkmm is `> 3.24.0`?

As I use Debian 10, I has `gtkmm3.0 (3.24.0-2)`

- https://packages.debian.org/source/buster/gtkmm3.0

JFTR, Actually ***only this SLVS*** file triggers segmentation fault — other SLVS files *SolveSpace* opens as expected.

P.S. As `$ solvespace *.slvs` still could be used for open it in GUI mode, I already tried to re-save this file — BUT *re-saved file also crashes* SolveSpace.

Symbian9, 2021-04-01:
Thanks @AlexanderP for sending me `solvespace-dbgsym_3.0.0~git1710-10ca307-1_*.deb`, here is backtrace log with `dbgsym` installed:

<details><summary>Backtrace Log (SolveSpace `3.0~10ca307`)</summary>

```Shell
user@user:~
$ gdb solvespace
GNU gdb (Debian 8.2.1-2+b3) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from solvespace...Reading symbols from /usr/lib/debug/.build-id/03/16c1010af7886af3059da7b2af8402e0f297f5.debug...done.
done.
(gdb) run
Starting program: /usr/bin/solvespace 
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffedaeb700 (LWP 5861)]
[New Thread 0x7fffed2ea700 (LWP 5862)]
SolveSpace!

Generate::ALL took 31 ms
[New Thread 0x7fffe6339700 (LWP 5863)]
[New Thread 0x7fffe59f7700 (LWP 5864)]
[New Thread 0x7fffe51f6700 (LWP 5865)]
[New Thread 0x7fffe49f5700 (LWP 5866)]
[New Thread 0x7fffdaeea700 (LWP 5867)]

(solvespace:5857): Gtk-WARNING **: 00:19:40.210: gtk_window_present_with_time() should not be called with 0, or GDK_CURRENT_TIME as a timestamp, the timestamp should instead be gathered at the time the user initiated the request for the window to be shown
[New Thread 0x7fffc260f700 (LWP 5870)]
[New Thread 0x7fffc1e0e700 (LWP 5871)]
[New Thread 0x7fffc160d700 (LWP 5872)]
[New Thread 0x7fffc0e0c700 (LWP 5873)]
[New Thread 0x7fffb3fff700 (LWP 5878)]
[New Thread 0x7fffb37fe700 (LWP 5879)]
[New Thread 0x7fffb2ffd700 (LWP 5880)]
[New Thread 0x7fffb27fc700 (LWP 5881)]
[Thread 0x7fffb37fe700 (LWP 5879) exited]
[Thread 0x7fffb3fff700 (LWP 5878) exited]
[Thread 0x7fffb2ffd700 (LWP 5880) exited]
[Thread 0x7fffc1e0e700 (LWP 5871) exited]
[Thread 0x7fffb27fc700 (LWP 5881) exited]
[Thread 0x7fffc260f700 (LWP 5870) exited]
Generate::REGEN (for bounding box) took 106 ms
Generate::REGEN took 204 ms
Generate::ALL (for bounding box) took 110 ms
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
Generate::ALL took 377 ms
Generate::DIRTY (for bounding box) took 102 ms
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
Generate::DIRTY took 238 ms
[... 32 lines cut ...]
#13 0x00007ffff7b4ccbc in Gtk::GLArea::~GLArea() () from /lib/x86_64-linux-gnu/libgtkmm-3.0.so.1
#14 0x00005555555ceaa7 in SolveSpace::Platform::GtkGLWidget::~GtkGLWidget (this=0x555555cb16e8, 
    __in_chrg=<optimized out>, __vtt_parm=<optimized out>) at ./src/platform/guigtk.cpp:443
#15 SolveSpace::Platform::GtkEditorOverlay::~GtkEditorOverlay (this=0x555555cb16c0, 
    __in_chrg=<optimized out>, __vtt_parm=<optimized out>) at ./src/platform/guigtk.cpp:610
#16 SolveSpace::Platform::GtkWindow::~GtkWindow (this=0x555555cb15e8, __in_chrg=<optimized out>, 
    __vtt_parm=<optimized out>) at ./src/platform/guigtk.cpp:728
#17 SolveSpace::Platform::WindowImplGtk::~WindowImplGtk (this=0x555555cb14e0, 
    __in_chrg=<optimized out>) at ./src/platform/guigtk.cpp:848
#18 __gnu_cxx::new_allocator<SolveSpace::Platform::WindowImplGtk>::destroy<SolveSpace::Platform::WindowImplGtk> (this=0x555555cb14e0, __p=0x555555cb14e0) at /usr/include/c++/8/ext/new_allocator.h:140
#19 std::allocator_traits<std::allocator<SolveSpace::Platform::WindowImplGtk> >::destroy<SolveSpace::Platform::WindowImplGtk> (__a=..., __p=0x555555cb14e0)
    at /usr/include/c++/8/bits/alloc_traits.h:487
#20 std::_Sp_counted_ptr_inplace<SolveSpace::Platform::WindowImplGtk, std::allocator<SolveSpace::Platform::WindowImplGtk>, (__gnu_cxx::_Lock_policy)2>::_M_dispose (this=0x555555cb14d0) at /usr/include/c++/8/bits/shared_ptr_base.h:554
#21 0x00005555555a6192 in std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0x555555cb14d0) at /usr/include/c++/8/ext/atomicity.h:69
#22 std::_Sp_counted_base<(__gnu_cxx::_Lock_policy)2>::_M_release (this=0x555555cb14d0) at /usr/include/c++/8/bits/shared_ptr_base.h:148
#23 0x00005555555a0fa3 in std::__shared_count<(__gnu_cxx::_Lock_policy)2>::~__shared_count (this=<optimized out>, __in_chrg=<optimized out>) at /usr/include/c++/8/bits/shared_ptr_base.h:1167
#24 std::__shared_ptr<SolveSpace::Platform::Window, (__gnu_cxx::_Lock_policy)2>::~__shared_ptr (this=<optimized out>, __in_chrg=<optimized out>) at /usr/include/c++/8/bits/shared_ptr_base.h:1167
#25 std::__shared_ptr<SolveSpace::Platform::Window, (__gnu_cxx::_Lock_policy)2>::operator= (__r=..., this=<optimized out>) at /usr/include/c++/8/bits/shared_ptr_base.h:1263
#26 std::shared_ptr<SolveSpace::Platform::Window>::operator= (__r=..., this=<optimized out>) at /usr/include/c++/8/bits/shared_ptr.h:335
#27 SolveSpace::SolveSpaceUI::Clear (this=0x5555558ee4c0 <SolveSpace::SS>) at ./src/solvespace.cpp:1023
#28 0x000055555559820b in main (argc=<optimized out>, argv=<optimized out>) at ./src/platform/entrygui.cpp:30
(gdb) 
```

</details>

rpavlik, 2021-04-20:
Hmm. So on a locally-built copy of SolveSpace, on Debian 10, latest master branch (3.0~0e0b0252) this loads fine through the GUI and file dialog, no crash. I have some stuff installed from backports but I'm reasonably sure nobody put gtkmm in backports, too intrusive...

robnee, 2021-04-25:
@rpavlik I wonder if we need to explicitly destroy or stop the timers when tearing down the app or when there is no geometry to display?  Maybe they are firing when we don't want them to.  I know for sure they don't always fire in a predictable order.

rpavlik, 2021-04-26:
Well, at least some of those mismatched new delete warnings about timers are false positives: shortly after I posted my last comment here, I found another bug report with a similar message and info on the root cause, so you'll notice there is now a comment on the relevant line of SolveSpace explaining it with a link to upstream 😁

It's possible there's a lifetime management thing going on here, but I wouldn't assume there is just based on my sanitizer results above.

Symbian9, 2021-04-26:
I just reinstalled system with MX Linux `MX-19.4` distro — on "clean install" *SolveSpace* `3.0~`https://github.com/solvespace/solvespace/commit/33091817738d780d904bcd3e077022d87ebf2cd3 can opens file without `Segmentation fault`.

1. Launch SolveSpace;
2. Open testcase file via "File > Open ...";
3. Close SolveSpace via "File > Exit"
4. Terminal log:

```
$ solvespace
SolveSpace!

json_object_from_file: error opening file /home/user/.config/solvespace/settings.json: No such file or directory

(solvespace:8064): Gtk-WARNING **: 21:45:58.476: gtk_window_present_with_time() should not be called with 0, or GDK_CURRENT_TIME as a timestamp, the timestamp should instead be gathered at the time the user initiated the request for the window to be shown
Generate::REGEN (for bounding box) took 107 ms
Generate::REGEN took 232 ms
Generate::ALL (for bounding box) took 121 ms
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
Generate::ALL took 353 ms
Generate::DIRTY (for bounding box) took 82 ms
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
bad, edge_inters=4
Generate::DIRTY took 198 ms
Generate::DIRTY (for bounding box) took 88 ms
Generate::DIRTY took 172 ms
```

@robnee @rpavlik, Does it looks OK?

robnee, 2021-04-26:
The messages look fine.  I would say if this file no longer seg faults then this issue can be closed.  I don't think anyone one else was able to reproduce.

closed 2021-04-26 by Symbian9

rpavlik, 2021-04-27:
yep, that's all good. Thanks for bearing with us as we dug into this one.