solvespace/solvespace #963: Hit assertion when open assembly

kalle, 2021-03-14:
### System information

- **SolveSpace version:** 3.0-36719c62
- **Operating system:** Fedora 33

### Expected behavior

I have an assembly (total.slvs in zip file), I edit one of the parts in the assembly and try to open the assembly again and I can see my changed part.

### Actual behavior

The assembly worked find, I edited a part (leg.slvs) and when I open the assembly solvespace hits an assertion:
```
File ./src/entity.cpp, line 614, function PointGetExprs:
Assertion failed: false.
Message: Unexpected entity type.
Aborted (core dumped)
```

### Additional information

Try to open total.slvs in the zip archive.
[coredump.zip](https://github.com/solvespace/solvespace/files/6137094/coredump.zip)

robnee, 2021-03-14:
If I remove this pt-on-face constraint from total.slvs then I can open the file again.  Do you remember which point you used to constrain the leg to the table surface?
  
```
Constraint.h.v=00000004
Constraint.type=43
Constraint.group.v=00000004
Constraint.ptA.v=8004004c
Constraint.entityA.v=80030043
Constraint.other=0
Constraint.other2=0
Constraint.reference=0
AddConstraint
```

kalle, 2021-03-16:
No. I tried to reproduce it with a couple different points but failed.