solvespace/solvespace #966: Test failure with -march=x86-64-v3

rpavlik, 2021-03-22:
From debian: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=983952

Build/test log is 
[solvespace_3.0.rc2+repack1-3_unstable_gcc10-v3.log](https://github.com/solvespace/solvespace/files/6185769/solvespace_3.0.rc2%2Brepack1-3_unstable_gcc10-v3.log)

Looks like here's the issue: one very small reproducibility issue:

```
test/constraint/equal_angle/test.cpp:5: FAILED: render doesn't match reference; 1 (0.00%) pixels differ
```

phkahler, 2021-03-24:
@rpavlik  Any idea why that test fails? Could the image created from xx86-64-v3 be usable to pass the test in all cases?

I've got a Zen (2400G) chip, but my GCC is not new enough to pass that option for answering these questions myself. I don't usually do Fedora Beta releases, but maybe I should update to 34Beta for this. Actually I haven't confirmed v3 is OK on zen1, all the info pairs specific instructions with the arch, not specific versions of chips.

rpavlik, 2021-03-24:
Not sure, this was an automated filing by the gcc maintainer in Debian.

phkahler, 2021-03-26:
I can reproduce it, but I can't see the difference in the test image. BTW the arch change gives it a notable performance boost on Zen. I'll look at the issue and measure performance more this weekend.

ruevs, 2021-03-26:
Probably this problem https://github.com/solvespace/solvespace/issues/565#issuecomment-603364397
@whitequark Mar 24, 2020 :
> Using nonstandard 80 bit floating point numbers is exactly the cause of the problem @nabijaczleweli is facing here. They're rounded differently than what the tests are expecting and so the test fixtures don't match.

So does the "xx86-64-v3" target use any new math hardware/instructions?

@phkahler can you attach the PNG from the failing test please? [Winmerge ](https://winmerge.org/) can compare PNGs visually - which is very convenient, I'll try to see where the difference is.

phkahler, 2021-03-26:
> So does the "x86-64-v3" target use any new math hardware/instructions?

It brings AVX and AVX2, so some new fused multiply-add instructions is all I see that might change things like rounding.

I'll post the PNGs later. I thought of it right after I deleted the failing one...

BTW in my day job I found that letting angles go greater than 2PI causes sin() and cos() accuracy to drop quite a bit at least on Intel hardware. I was producing variable frequency sine waves by adding dt*omega at each time step and things got bad until I added code to wrap around at 2PI. I need to check if Helix wraps around - I don't think it does.

phkahler, 2021-03-26:
@ruevs 
Here is the original test image
[image]

And here is the one produced under march=x86-64-v3
[image]

ruevs, 2021-03-26:
There is a one pixel difference in the length of the horizontal line of the right angle constraint:
[image]
Here the area is marked with a square:
[image]
[image]

phkahler, 2021-03-27:
@ruevs I don't know why there are so many variants of this test in the folder. I could try some silly things like just moving the geometry around a little and seeing if I can find a variation that passes on both architectures. It seems to be a fluke since all the other tests pass.

closed 2021-04-03 by phkahler