solvespace/solvespace #967: Export of normals & datum points not works for PDF and SVG

Symbian9, 2021-03-23:
### System information

- **SolveSpace version:** 
- **Operating system:** 

### Expected behavior

All objects, which styles are marked `[v] export these objects`, should be exported in PDF/SVG/etc, including *normals* & *datum points**.

### Actual behavior

Actually *normals* & *datum points* does NOT exports to PDF & SVG, even ALL styles set to `[v] export these objects`.

@robnee, I guess it related to: https://github.com/solvespace/solvespace/pull/946

### Additional information

- ***SolveSpace*** (`slvs3-test.slvs`)
[image]

- ***qpdfview*** (`slvs3-test.pdf`)
[image]

- ***Inkscape*** (`slvs3-test.svg`)
[image]

robnee, 2021-03-23:
Datum points and normals are not mapped to edges/Beziers when exporting to vector formats.
```
        default:
            // Not a problem, points and normals and such don't generate curves
            break;
```

I will look into the color of export construction entities.

robnee, 2021-03-24:
I think I found the problem with the export of construction lines (in 3d).  It looks like construction entities and constraints (and maybe others) don't explicitly set their styles to s002-#def-construction s006-#def-constraint.  They are implicitly styled.  This means that code that renders or exports them needs to derive the style using  `Style::ForEntity(h)`  `Entity::GenerateBezierCurves` is already doing this but `Entity::GenerateEdges` is not

To fix the lack of vector exports for points and normals we'd have to render them to a `GetEdgesCanvas` as is done when exporting constraints.  The code calls this "an awful temporary hack" so I'm not sure if it's a good idea or not.

Symbian9, 2021-03-24:
> I think I found the problem with the export of construction lines (in 3d). I'll do a PR for this.

Great. Just do not close this issue, as fix of construction lines export is a half of this issue.

> The code calls this "an awful temporary hack" so I'm not sure if it's a good idea or not.

As for vector exports for points and normals — it should work as expected, i.e. if they are visible in main window & its stile set to `[v] export these objects` they should be exported.

@jwesthues @phkahler, what you think would be a solution to fix this?

robnee, 2021-03-24:
@phkahler I can implement this for normals using `GetEdgesCanvas` since that already implements `DrawLine` and that's all that's needed to render normals.  I'm not sure how to implement this for datum points though.  What I assume I need to do is something like render a tiny filled square polygon but  oriented normal to the plane of the screen/camera so that when its projected into 2d for export it's not distorted.  it's tricky though because while the manual hidden edge removal used for export is successfully clipping the fully obscured datum points it's also clipping some of the edges that should be unobscured.  You can see it if you look closely in the image below.

[image]

Symbian9, 2021-03-25:
> render a tiny filled square polygon but oriented normal to the plane of the screen/camera so that when its projected into 2d for export it's not distorted

@robnee Also, take a look on export, but also `pts-coincident` constraints dots as:

-  datum points - *tiny filled square polygon*
- `pts-coincident` constraints dots - *tiny filled circle/ellipse polygon*

Another solution is to export *datum points* and  `pts-coincident` as "as is" just with some styling (i.e. like point with defined size) according SVG  and PDF specifications.

JFTR, Maybe, in SolveSpace we need to redesign *datum points* & `pts-coincident`

> it's tricky though because while the manual hidden edge removal used for export is successfully clipping the fully obscured datum points it's also clipping some of the edges that should be unobscured. You can see it if you look closely in the image below.

As for *visible datum points*, in PDF/SVG export it just should be rendered 2nd level layer in level orders:

- 1st layer - Constraints (top level - above all);
- 2nd layer - Visible datum points (behind constraints, but above all other) 
- ...
- N-1 layer - Visible edges (separated in)
- N layer (bottom layer) - Visible color fills (i.e. visible shaded triangles)

Guess its need a little work on ***layers order*** design for SolveSpace (maybe need review some of them)

As for me, shaded triangles always should be behind all other layers.

phkahler, 2021-03-25:
@robnee  To do the little squares correctly you'd have to check if the "point" is visible and then save it for drawing on top of everything else - on another layer so to speak - rather than allowing parts of the entity to be occluded. Or make a separate pass after everything else is drawn. I'm not sure how useful this is, but it would make it more consistent, and you have been working in the styles.

robnee, 2021-03-25:
@phkahler @Symbian9 Thanks.  I agree with both of your comments in principle but it's the implementation where I get stuck.  The vector exports in 3d all work by first converting all entities and constraints to edges where possible.  Datum points are skipped because they are zero-length edges.  Then `ExportLinesAndMesh` takes this and does occlusion testing to remove hidden edges, splitting and recombining edges as necessary.  The resultant set of vectors are then mapped to the export format's representation of edges.  What I did above was convert the datum points to a quad of 4 edges each so I could feed them into `ExportLinesAndMesh`. You can see that the occlusion works perfectly, splitting these little edges where they are behind or embedded in the object.  The one completely occluded point is indeed completely removed.  Except this clipping isn't what we want for datum points.

There is really no concept of layers here.  Constraints are a special case, they are never hidden-edge-removed so they are effectively in a layer on top of the drawing.  Everything else is in a single layer.  This won't work for datum points, some are occluded and some aren't.  If there is a way, independent of the `ExportLinesAndMesh` to determine if a datum point in the view is occluded or not then they _could_ be layered on top after the edges are processed.  This really starts to look like a rewrite of the vector export and not just an enhancement though.

robnee, 2021-03-25:
Separately, the issue with styling of construction entities in 3d is indeed related to #946 and is a one line addition.  I'll do a PR for that while we consider the datum point issue.