solvespace/solvespace #969: Should we fork mimalloc?

phkahler, 2021-03-24:
There are 3 commits to mimalloc that we really need in SolveSpace. The lead maintainer indicated just over a month ago that he'd make a new release "this week" but it's been a month and his github activity has dropped off the last few months. Should we fork it and tag the latest version on the dev-slice branch (which works for us AFAICT) so we can get our version 3 out?

ruevs, 2021-03-25:
Why not simply point the external to the specific commit that works for us for now?

phkahler, 2021-03-25:
> Why not simply point the external to the specific commit that works for us for now?

Can we do that without it having a tag? Should we? it's not even on their main branch.

ruevs, 2021-03-26:
> > Why not simply point the external to the specific commit that works for us for now?
> 
> Can we do that without it having a tag? Should we? it's not even on their main branch.

- Can we do that?
Of course we can - it is trivial.
https://www.git-scm.com/docs/gitsubmodules#_workflow_for_a_third_party_library

>since submodules are fixed to an arbitrary version, the other project can be independently developed without affecting the superproject, allowing the superproject project to fix itself to new versions only when desired.

- Should we do it?
Why not? Checking out a particular branch/revision on the submodule seems (to me) much simpler and cleaner than forking it. And when they release a version we "like" we just update the submodule.

phkahler, 2021-03-31:
@ruevs can you point it to their dev-slice branch which is 200 commits ahead of master? I'm currently quarantined away from my dev box :-(

ruevs, 2021-04-03:
> @ruevs can you point it to their dev-slice branch which is 200 commits ahead of master? I'm currently quarantined away from my dev box :-(
> Never mind.

Sorry just saw this now :-(

closed 2021-04-03 by phkahler