solvespace/solvespace #970: Poor rendering quality far away from the origin

baryluk, 2021-03-25:
- **SolveSpace version:** 3.0.rc2+repack1-3
- **Operating system:** Debian testing

[image]
[image]

Some points, are not even points.

[sol.zip](https://github.com/solvespace/solvespace/files/6201213/sol.zip)

Sure, the unusual mix of big and small values is problematic, but I don't think it should cause so big issues as they show here, especially related to rendering.

It almost feels like all the computations are done in single precision numbers, instead using doubles.

For rendering, when zooming in, one of the solution for better GPU rendering is, to translate and scale close to origin and use that for rendering. For clicking, the transformation can be then reversed. That should improve quality, even if GPU is doing calculations in single precision or even below it. But that would require generating vertex buffers on position changes. No idea if / how that is expensive or not.

robnee, 2021-03-25:
Are you talking about the circles rendered as piece-wise-linear-segments (vs smooth curves) or the poorly rendered line segments?  If the latter what does the GL vendor/renderer/version at the bottom of the configuration window say?  I know there are some rendering issues with some Intel GPUs.

baryluk, 2021-03-26:
>Are you talking about the circles rendered as piece-wise-linear-segments (vs smooth curves) or the poorly rendered line segments? 

The second part.

Antialiasing quality is very poor for some lines, some almost vanish, or have significant moira-like pattern.

Some small squares, are rendered as rectangles.

[image]

```
$ glxinfo  -B
name of display: :0
display: :0  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: AMD (0x1002)
    Device: AMD Radeon (TM) R9 Fury Series (FIJI, DRM 3.40.0, 5.10.0-4-amd64, LLVM 11.0.1) (0x7300)
    Version: 20.3.4
    Accelerated: yes
    Video memory: 4096MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 4.6
    Max compat profile version: 4.6
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.2
Memory info (GL_ATI_meminfo):
    VBO free memory - total: 3784 MB, largest block: 3784 MB
    VBO free aux. memory - total: 3993 MB, largest block: 3993 MB
    Texture free memory - total: 3784 MB, largest block: 3784 MB
    Texture free aux. memory - total: 3993 MB, largest block: 3993 MB
    Renderbuffer free memory - total: 3784 MB, largest block: 3784 MB
    Renderbuffer free aux. memory - total: 3993 MB, largest block: 3993 MB
Memory info (GL_NVX_gpu_memory_info):
    Dedicated video memory: 4096 MB
    Total available memory: 8192 MB
    Currently available dedicated video memory: 3784 MB
OpenGL vendor string: AMD
OpenGL renderer string: AMD Radeon (TM) R9 Fury Series (FIJI, DRM 3.40.0, 5.10.0-4-amd64, LLVM 11.0.1)
OpenGL core profile version string: 4.6 (Core Profile) Mesa 20.3.4
OpenGL core profile shading language version string: 4.60
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 4.6 (Compatibility Profile) Mesa 20.3.4
OpenGL shading language version string: 4.60
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile

OpenGL ES profile version string: OpenGL ES 3.2 Mesa 20.3.4
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.20

```

ruevs, 2021-03-26:
Yes, I have also noticed this (a long time ago). For example on a Raspberry Pi where the OpenGL renderer seems to be a 128bit software one the problem does not exist:
https://github.com/solvespace/solvespace/pull/825#issuecomment-734899712