solvespace/solvespace #972: Performance under different build options

phkahler, 2021-03-27:
### System information

- **SolveSpace version:** 3.0~3dd2fc00
- **Operating system:** Fedora 34 Beta
- **CPU:** Ryzen 2400G - system Mellori ITX
- **RAM:** 16GB but SolveSpace only uses about 300MB

### CMAKE options: 
(option 0 is always enabled)

0) -DCMAKE_BUILD_TYPE=RelWithDebInfo
1) -DENABLE_OPENMP=yes
2) -DENABLE_LTO=yes
3) -DCMAKE_CXX_FLAGS=-march=x86-64-v3

|OMP| LTO| 64v3| TIME (s)
|----|----|----|----
|    |    |    | 21.077
|    |    | [x] | 21.550
|    | [x] |    | 18.080
|    | [x] | [x] | 18.271
|[x] |    |    | 6.371
|[x] |    | [x] | 6.552
|[x] | [x] |    | 5.751  5.468
|[x] | [x] | [x] | 5.498  5.536

Procedure was to open perf.slvs then change the number of repeats in the last group from 4 to 40. Do this 3 times for each set of build options and take the average of the 3. The last two I did 3 runs twice because they were close and after the first run this was the only time x86-64-v3 was better than without it. The second set of numbers is more consistent with the others, as is the average of all 6 runs.
[perf.zip](https://github.com/solvespace/solvespace/files/6214890/perf.zip)

### Conclusion:
The biggest improvement by far is to enable OpenMP. An additional speedup is available with LTO but will increase your build time. Using x86-64-v3 yeilds a tiny regression on this test with this processor which is a Zen+ so it may be worth using this option if other areas of the code benefit from it, or on newer Intel or AMD chips.

It is also notable that when I started using this file as a performance test (someone posted it ot the forum), it was taking about 3 minutes, or 58 seconds with OMP. From 180 seconds reduced to 6 is an improvement of 30x, and I don't think that includes all the performance improvements since version 2.3.  It does include my switch from Fedora 32 to 34B and GCC 11 which I did not think to take measurements before/after.

phkahler, 2021-04-21:
Can someone try this on an Apple M1? @vespakoen ?  Just the regular OpenMP and LTO for comparison would be nice.

I guess running the x86 version would be easy for anyone, but SolveSpace can be compiled native to the M1 as well.

vespakoen, 2021-04-21:
With a recent stable build I got 1.285 / 1.300 / 1.249

phkahler, 2021-04-21:
@vespakoen  If that's at default chord tolerance (0.1 percent) then I'm very impressed. My results were on a 4 core / 8 thread Amd Zen+.  You have twice the cores, but that would get you half the time at best, and you're much faster than that. I have not seem a head to head comparison of the M1 to any Intel or AMD until now.

ruevs, 2021-04-21:
> I have not seen a head to head comparison of the M1 to any Intel or AMD until now.

https://www.youtube.com/watch?v=l1j-aF_wyzU

phkahler, 2021-04-23:
I just reran 3.0 final with OpenMP and LTO.  5.502 / 5.503 / 5.552  with the default CT of 0.1 percent. I don't want to change the CT, but until @vespakoen confirms I wanted to see if increasing it can get anywhere near his M1 results.  Using the old default of 0.5 percent I can bring the time down to just over 4 seconds, so that's not a huge deal for this test but it still makes a difference.

vespakoen, 2021-04-26:
Looks like I use 0.1 indeed...

[image]

I just ran it again, and got different results:

```
1. 2769
2. 2779
3. 2813
4. 2759
```

Not sure what changed between this run and the last time I ran it... same executable...

phkahler, 2021-04-26:
Oh, dont use triangle mesh. When I keep it NURBS there are 3 holes that get red edges. That's OK. Looks like the M1 is quite impressive :-)

phkahler, 2021-09-04:
I just saw that GCC is going to enable some vectorization with optimization level -O2.  So looking at our compiler flags it seems releases are built -O2.  Maybe if we were building at -O3 the 64v3 flag may have been useful. This may also explain why some things I tried a while back didn't seem to have the expected result.