solvespace/solvespace #977: Rendering bug when some objects are transparent

matt-hammond-001, 2021-03-28:
### System information

- **SolveSpace version:** Version 3.0 (3.0~36719c62)
- **Operating system:** Mac OS 10.15.7

### Expected behavior

One object, with opacity 0.5 is infront of another object. The portions of that second object that are behind the semi-transparent object should appear the same as the portions that are not occluded.

### Actual behavior

The portions of an object occluded by a semi-transparent object appear rendered incorrectly. It looks like only the faces that are facing away are rendered, instead of the faces facing towards the camera.

### Additional information

Example file, and example screenshot attached.
[rendering transparency bug.slvs.zip](https://github.com/solvespace/solvespace/files/6217584/rendering.transparency.bug.slvs.zip)
[image]

ruevs, 2021-03-28:
This is a duplicate. Here is the original issue: https://github.com/solvespace/solvespace/issues/522

Symbian9, 2021-03-28:
> Here is the original issue: `#522`

I suppose that `#522` should be reopened as it has more details on SolveSpace transparency issue.

ruevs, 2021-03-29:
> > Here is the original issue: `#522`
> 
> I suppose that `#522` should be reopened as it has more details on SolveSpace transparency issue.

It was closed with "wontfx" and a pretty decent explanation by Whitequark why not. If we reopen it all we'll have is one more open issue.

If we are to keep one open that one is definitely better, but I'd rather clese this one as well. Otherwise we will simply have one more open issue. @phkahler what do you think?

Symbian9, 2021-03-29:
> It was closed with "wontfx" and a pretty decent explanation by Whitequark why not.

I suggest that `wontfix` tag, added by ex-maintainer personally, should be removed from `#522` as transparency issue is a serious *SolveSpace* `bug` which should be fixed *better late than never*.

> If we are to keep one open that one is definitely better, but I'd rather close this one as well. Otherwise we will simply have one more open issue.

I'm fully agree on closing `#977` as `duplicate` and reopening `#522` for continuous discussion on transparency `bug`.

matt-hammond-001, 2021-03-29:
Thanks - sorry I didn't spot #522 ... I searched a little but somehow didn't find it.

phkahler, 2021-03-29:
I'd say reopen the other one. It's an obvious rendering error even if nobody can fix it any time soon. Maybe leave the tag? ;-)

Symbian9, 2021-03-29:
> Maybe leave the tag? ;-)

Just remove `wontfix` tag and add `help wanted` tag instead.

closed 2021-03-29 by ruevs