solvespace/solvespace #981: Three Sphere Performance Bug

giannissc, 2021-03-29:
I am posting here about a potential performance bug in solvespace. The issue is easy to reproduce. Just create a sphere and then create a translate group. This makes solvespace unusable for a few minutes. I don't this operation should trash performance so much. Just leaving it here so that you can investigate

pjanx, 2021-03-29:
Would you attach a file that shows the problem?

pjanx, 2021-03-29:
I can't reproduce this even with 500 spheres: [hundreds-of-translated-spheres.zip](https://github.com/solvespace/solvespace/files/6224401/hundreds-of-translated-spheres.zip)

phkahler, 2021-03-29:
What version of SolveSpace?
What is your chord tolerance set to?

giannissc, 2021-03-30:
I actually discovered what the problem is by seeing your example @pjanx. Instead of using half a circle for generating the sphere I used a full circle.

giannissc, 2021-03-30:
Using half a circle is night and day in performance

closed 2021-03-30 by phkahler