solvespace/solvespace #984: SolveSpace Crashed  when I draw a line

zhulaojianke, 2021-04-01:
### System information

- **SolveSpace version:**3.0~36719c62. 

- **Operating system:**  Win7 Ultimate Edition sp1

### Expected behavior

Only draw a line

### Actual behavior

It pops up a message window.
[image]

### Additional information

Sorry, I can't give you any files. When I click to generate a debug report, the solvespace quits itself.

zhulaojianke, 2021-04-01:
It seems only occurs on chinese interface, when I change the language to english, there is nothing go wrong.
[image]

zhulaojianke, 2021-04-01:
When I close the "check sketch for closed contour", this crash also be avoided, even in chinese interface.

phkahler, 2021-04-01:
@zhulaojianke  The chinese translation has not been well tested. When I merged the pull request all I did was verify that the menus switch to chinese, which I can't read. It never occurred to me that a "bad" character might crash the program.

This is really easy to recreate - draw a single line segment with language set to chinese.

Sounds like it may be an issue with the on-sketch vector font not handling something correctly. We may need to change the chinese text for the "not a closed contour....." message.  Unfortuantely the submitter has not responded since the translation was submitted.

If you want to help that would be great, otherwise I'll try deleting the chinese version of that message and leave it english. That's better than crashing!

zhulaojianke, 2021-04-02:
I don't think it's a translation problem，it seems to be a character encoding problem, because when I put a comment on the workplace and input chinese character such as “试试看”, it also return a wrong display.
[image]
It seems any chinese character will be displayed wrong here.
To display chinese character correctly, I suggest to use these character encoding, such as [GB18030]简体中文（UTF-PRC）, [GBK]Windows Chinese Simplified（CP 936）or GB-2312, [MACCHINESESIMP]MacChineseSimp.

I don't know how to program or sinicalize, and I feel that's all I can do to help. I'm so sorry.

phkahler, 2021-04-02:
@zhulaojianke  The problem is the on-sketch text display uses a special vector font. We can't change the font or encoding without making huge changes to how text is drawn on the sketch. Many characters are missing from that font and they usually display as the same symbol. Looking at the translation file I see the following:

not closed contour, or not all same style!
未闭合轮廓，或样式不一致!

If I copy those chinese characters into a comment constraint SolveSpace will crash (even in english). If I only copy the first part up to the comma, it will not crash but only 2 of the symbols are shown correctly with the "missing glyph" showing for the other 3.

I'm happy to revert this one string to english, or change it to something else chinese if you can find a suitable text that will display properly (try entering text in a comment constraint.  Then post that text here on github and I'll update it).

zhulaojianke, 2021-04-03:
OK! I find it! Replace the SBC case comma to a DBC case comma, or do not use SBC case punctuation mark in the alarm message or comment.
For example, I input the same message using DBC case comma in the comment
[image]
and then
[image]
It no longer crashes, although it still didn't display correctly.

zhulaojianke, 2021-04-03:
I have tried other inputting contained SBC case punctuation mark, the SolveSpace crashed too.
For example, I input these using SBC case comma, and press enter key
[image]
and then...
[image]

phkahler, 2021-04-03:
@zhulaojianke  there are some characters missing from the font. Can you find a way to translate the message that makes sense, doesnt crash solvespace, and displays correctly?

phkahler, 2021-04-03:
I am closing this issue after changing the comma. The message no longer crashes.

closed 2021-04-03 by phkahler