solvespace/solvespace #986: Ambient 1.0 settings ignored on PDF and SVG export

Symbian9, 2021-04-01:
### System information

- **SolveSpace version:** `3.0~` https://github.com/solvespace/solvespace/commit/10ca307d986ab0c52faaab4fcaedb26bd42d1fd6
- **Operating system:** Debian 10 buster

### Expected behavior

Export of shaded triangles should depend on `ambient lightning` settings:

- Special use-case: [b]if[/b] `ambient lightning` set to `1.0` *shaded triangles should be exported as "flat color"*;

@robnee, See *"Flat colors render mode"* thread:

- https://github.com/solvespace/solvespace/issues/524

### Actual behavior

Actually if set `ambient lightning` to `1.0` — triangles with same color value (according solid group color) are NOT rendered as in pure "flat colors" (see PDF and SVG in attached ZIP).

In same time if *"File > Export Image..."* — it shown exactly as expected (see PNG file in attached ZIP).

### Additional information

robnee, 2021-04-01:
@Symbian9 Just so I understand, if ambient light == 1.0 then don't compute the lighting and just use the group's solid color for all triangles?

phkahler, 2021-04-01:
@robnee For on-screen rendering they are additive. You get flat coloring by setting ambient = 1.0 and set the other lights brightness to 0.0.   It's possible to use both and I don't recall any clamping if things get too bright (maybe should ;-)

phkahler, 2021-04-01:
I should clarify that. It's ambient*(r,g,b) + c1*(r,g,b) + c2*(r,g,b)

Where (r,g,b) is the surface color, c1 and c2 are coefficients for light 1 and 2 which are simply the dot product of the light direction and the surface normal (snells law). Apparently I didn't add the ambient term to 2D export - nor did I know where that would go ;-)

robnee, 2021-04-02:
Looks like the lighting value needs to be clamped prior to applying it to the rgb values.  

There is also a bug on the config screen for the ambient lighting which doesn't manifest for me on Linux but does on Windows.  It looks like the Printf is copied from the light direction but needs to have the arg for the light number removed.  This caused crazy behavior when I was testing with Windows.  I'll include in the PR.

PR #988 (2021-04-02): Clamp lighting value on export to fix flat colors
This closes #986.  The computed lighting value when ambient lighting value == 1.0 needed to be clamped to 1.0 in order to achieve flat shading as expected.

Also, this fixes the Printf format for the ambient lighting entry on the configuration screen.  This was causing strange behavior when attempting to edit the ambient lighting value on Windows.  Oddly this didn't seem
to affect Linux.
closed 2021-04-02 by phkahler