solvespace/solvespace #996: how to step dimension the rotation with the angle great than 180

zhulaojianke, 2021-04-04:
### System information

- **SolveSpace version:** 3.0~ 312a8df1
- **Operating system:** Win7 Ultimate Edition sp1

During I learn Solvespace tutorial No.4 https://solvespace.com/linkage.pl , at the last step, I use the "Step Dimension" to drive the line rotate like a motor, I input an angle 360 at the "finish"  box, and then click “step dimension now”, and then **the line is not rotate a full circle! but rotate in the opposite direction when the angle reach 180!** . I tried it many times, it only rotates between 0 and 180 unless I pull that line over the X-axis with my mouse. 

I don't know how anyone else has solved this problem.

I lookup this issue at this site, I've noticed that other people have this problem as well. It seems that to solve this bug is very hard.

I suggest to use radians instead of angles, but Solvespace is not support inputting radian, there is no switch of the radians and angles.  I hope this method could solve this problem.

Or please tell me how to simulate the rotation with the angle great than 180 in solvespace?

Symbian9, 2021-04-04:
> the line is not rotate a full circle! but rotate in the opposite direction when the angle reach 180! . I tried it many times, it only rotates between 0 and 180 unless I pull that line over the X-axis with my mouse.

That is known issue (not solved yet).

> how to step dimension the rotation with the angle great than 180

If expected rotation is between `> 180°` and `< = 360°` it's possible to use a trick: use *arc* length equal to *line* and step dimension its length.

zhulaojianke, 2021-04-04:
thanks, good idea! :)

zhulaojianke, 2021-04-21:
I have find a better method to solve this problem. :)
[Angle ratio.zip](https://github.com/solvespace/solvespace/files/6352976/Angle.ratio.zip)

And I hope you could add "Angle Ratio" and "Angle difference" commands in the "constrain" menu, just like "length ratio" and "length diffrerence" commands. These two commands are very useful for driving the "motor" turn many rounds in step dimension mode.

Symbian9, 2021-04-21:
> I have find a better method to solve this problem. :)

Every SolveSpace trick has its pros'n'cons ;)

> And I hope you could add "Angle Ratio" and "Angle difference" commands in the "constrain" menu, just like "length ratio" and "length diffrerence" commands. These two commands are very useful for driving the "motor" turn many rounds in step dimension mode.

Create new issue thread for this feature request.

zhulaojianke, 2021-04-22:
> Every SolveSpace trick has its pros'n'cons ;)
>
>
> Create new issue thread for this feature request.

Haha, All right, I'll take your advice.

I have watched all your challenge videos on youtube, really good! And can you guess how I made the shape below? :D
[image]

Using the simple functionality of solvespace to complete some challenges makes me feel a sense of accomplishment. :)

ruevs, 2021-04-22:
> And can you guess how I made the shape below? :D

"Step Translating" ? ;-)

Symbian9, 2021-04-22:
> And can you guess how I made the shape below?

Yes, I can. Even more, I can find more than one solution for such solid creation ;D

BTW, Such sort of showcase is out of scope of current issue thread, so continuous discussion on this showcase should be moved to [/r/SolveSpace](https://old.reddit.com/r/solvespace), [Twitter](https://twitter.com/hashtag/solvespace), etc.

zhulaojianke, 2021-04-22:
Oh, I'm sorry. I'll take care of that in the future.