solvespace/solvespace #999: Locales switching does NOT work for "Property Browser"

Symbian9, 2021-04-04:
### System information

- **SolveSpace version:** `3.0~` https://github.com/solvespace/solvespace/commit/2df786e7415501c748f34741aa1d5c561e730814
- **Operating system:** Debian 10 buster

### Expected behavior

1. Switching locales should take effect not only *main window*, but also *"Property Browser" window*;
2. Remember selected locale for next SolveSpace session.

### Actual behavior

1. Switching locales should actually takes effect only *main window* — *"Property Browser" window* instead always is English-only;
2. Selected locale does not used for next startup — instead each SolveSpace session starts in English.

### Additional information

<details><summary>Screenshots (taken after each locale switch)</summary>

[image]

[image]

[image]

</details>

phkahler, 2021-04-04:
The property browser is not coded in a way that allows localization. I agree that it would be an improvement, but it will have to wait for a rewrite of all that code. There are no specific plans to do that yet.

phkahler, 2021-04-04:
Duplicate of #319

closed 2021-04-04 by ruevs

ruevs, 2021-04-07:
In fact the even older issue is https://github.com/solvespace/solvespace/issues/112