solvespace/solvespace #1006: Feature request: Navigation Cube

giannissc, 2021-04-07:
I was recently trying to create a piece (linked [here](https://github.com/solvespace/solvespace/files/6272427/dove-table.zip)) using slanted extrusions. I wanted a 45° relative to the front view to create a new workspace to push the extrusion through. I was able to do that by changing the projection parameters inside property browser -> view but it would have been so much easier to do so with a graphical solution similar to [FreeCAD's navigation cube](https://wiki.freecadweb.org/Navigation_Cube), [Fusion360's ViewCube](https://www.autodesk.com/research/publications/viewcube) or [Blender's Navigation Gizmo](https://docs.blender.org/manual/en/latest/editors/3dview/navigate/introduction.html#navigation-gizmo)

It could also improve usability and help with other issues such as (#123, #888, #325) nagivation and it relates to #956. It could potentially make changing the current projection parameters move discoverable and intuitive to use

ruevs, 2021-04-07:
Are these "navigation cubes" just hints for the orientation of the model?
If yes I think we have enough of those:
[image]

If they actually allow you to "set" specific orientations - what is the user interface? Sorry I have not used any of those.

We have `W`, `F2` and `F3` (`View` menu) for orientation presets.

Usually the best way to create work planes in odd orientations is to link them to some construction elements (Sketch | Anywhere In 3d) constrained in the orientation you want. The advantage is that if you re-orient the construction elements (e.g. two line segments) the work plane linked to them also changes orientation.

giannissc, 2021-04-07:
I know there are plenty of ways to do that and they are. I am just think it will be easier to do for some standard value e.g. 45° (left or right side of current view) or quickly switching between different angles e.g. front view to bottom back view

giannissc, 2021-04-20:
Here is how I imagine a navigation cube in solvespace would be:

It would be positioned where the existing axes are like so:
[image]

This is my first design iteration. In the next pictures I also color coded the axes to make it easier to control.
Here is again how it would look in isometric view without the axes
[image]

And finally how it would look in an orthographic view:
[image]

giannissc, 2021-04-20:
What I like about this design is that the arc give you a more intuitive way to control the screen projection. Being in an orthographic view a pressing on the arc would allow you to specify the angle of the drawing (in a similar way you can do using Ctrl Click). However, being that the arc and the letters are extruded it gives you access to the other dimensions as well.

Now say I wanted to rotate from the front view so that half my screen show the Front and the other half show the Right view I can just click at the green arc and say take me to the 45°

If I wanted to jump to the right view I just click the red letters in the middle

giannissc, 2021-04-20:
And while I am thinking of specifying pitch, yaw and roll angles using text input they can also be made interactive

ruevs, 2021-04-20:
In think #1017 and this are closely related.

As for me I might be a masochist but the current options for orienting the view are fine for me. This looks like "UI fluff" but if someone implements it cleanly as an option - fine. Personally I don't want this cube in the screen all the time.

giannissc, 2021-04-21:
Yeah I agree it is not going to change the way solvespace currently works. The current navigational features are sufficient. What I think this will do is make some edge cases easier and be make the UI a bit more friendly to beginners

phkahler, 2021-04-21:
@giannissc I'm with @ruevs in thinking it's not needed, but if someone implements it cleanly as an option we should accept it. It's probably more about familiarity for people coming from other tools.

ruevs, 2021-04-22:
Aaaand this is a duplicate after all ;-) https://github.com/solvespace/solvespace/issues/70

Thovthe, 2022-04-15:
Close this in favor of #70?

closed 2022-04-15 by ruevs