solvespace/solvespace #1007: Feature request: Supress solid model in property browsers

giannissc, 2021-04-07:
There is currently the "shown" check box at the top level of the property browser that activates/deactivates features generated by the different shapes/solid such as normals, points, line segment etc. It would also speed a lot of workflows if next to the "shown" columns there was a "solid" column that could toggle the solid model on/off. The current way to do that is to go to the solid group's model and choose "suppress solid model" but I think it would improve usability to be able to do that right at the top of the property browser rather than navigating inside the different groups.

Usecase: I find myself often trying to turn off a solid model when working on the inside of a shape. I will usually draw something re-enable the solid model to see how it looks and if there are any errors or imperefections disable that solid model again and continue working with the sketch.

[Example clip.zip](https://github.com/solvespace/solvespace/files/6272678/clip.zip)

Drawback: The only drawback that I see is that it kind of goes against solvespace philosophy. All operations appear to be generic  and suppressing the solid model will only work with 3D groups. In those cases there oculd just be a dash indicating that there is no model to suppress.

ruevs, 2021-04-07:
Are you aware of the rightmost icon in the Property Browser tool bar? It is a three state one controlling "visibility".
I find that by using it and the check boxes in the "shown" column I practically never have to disable solid models for the purposes of drawing something "inside" or occluded by them. 

[image] [image] [image]

giannissc, 2021-04-07:
I know that but I never actually used it when desiging. I will give it a go 😄

someSven, 2021-04-18:
I had the same thought than OP and wasn't using that icon as well. Somehow escaped my memory. It seems to work pretty well.
However, I also wonder, if deactivating the rendering of some parts of the model could cut down on resource usage while working on some part. This might be an edge case, though, for users using some weak machine to not procrastinate by using the stuff on their bigger computers (games, etc).

ruevs, 2021-04-19:
Deactivating the rendering will not help much in most cases.

giannissc, 2021-04-20:
I have actually had time to play around with it and the buttons at the top really help a lot. The only use case I have had for the solid model button has been in another program, Fusion360. It came in really handy when importing step assemblies. I wanted to copy the designs to solvespace and deactivating some component made it easier to work with that larger model to design and assemble mine in solvespace. Other that it might be convenient and more obvious to deactivate some solid models I cannot think of any other reason why these button will need to be exposed at the top level. Feel free to close 😄

giannissc, 2021-05-08:
I have actually found a usecase for this feature. I have found that it would be useful to be able to have quick access to the solid model show/hide when trying to export different parts of the design for manufacturing. Some surfaces when occluded by other objects around might be hard to select for example

giannissc, 2021-05-14:
Yeah that seems resonable to me

hinell, 2022-01-27:
Thumbing up. I just found myself contantly refreshing flag for supressing solids... Especially when I've a lot of unions or differences.