solvespace/solvespace #1009: Is the Bezier Spline tool only allowed users to draw up to 10 points?

zhulaojianke, 2021-04-11:
I just used the "Sketch Cubic Bezier Spline" tool to draw a 2D CAM model, and it seemly only permit me draw ten points once, and The problem caused my CAM curve can not be closed. 

I wish that the developers could allow users to draw more curves points than 10.  :)

phkahler, 2021-04-11:
You can draw two or more curves with shared endpoints. You may also constrain them to be tangent at the ends.

zhulaojianke, 2021-04-11:
I find a way to solve this problem, use several point to define a spline circle and right click on it, select "Add spline point".

And then there are 4 new problems generated:
(1) Can't define a tangent constrain between tow shapes, such as a line and a circle, two circle, a circle and a spline curve, etc. Then it seems hardly to define a CAM contact relation.
(2) Can't define a perpendicular constrain between a line and a spline curve.
(3) Can't constrain a point on a spline curve (not endpoint).
(4) Can't combine two or more lines or curves that share endpoints in one group, to generate a new single shape or curve, just like weld. For example, just use the method you ever said to generate a closed contour by two seperate splines, even they were tangent at the ends, when a circle roller is tangent to this closed contour (assume that problem (1) can be solved), the roller can only slip or scroll on one curve, but can not do this on the whole contour, because the contour is not one single shape or curve. Do you understand what I mean? Sorry my english so poor.

I have searched example of CAM kinematics simulation using solvespace online, but found no answer...

zhulaojianke, 2021-04-11:
Relvevant worksheet.
[CAMproblems.zip](https://github.com/solvespace/solvespace/files/6292471/CAMproblems.zip)

robnee, 2021-04-11:
> (1) Can't define a tangent constrain between tow shapes, such as a line and a circle, two circle, a circle and a spline curve, etc. Then it seems hardly to define a CAM contact relation.

You can define a line tangent to a circle with the aid of a construction line through the center and perpendicular to the line.  A similar technique will let you do a line perpendicular to a circle.

zhulaojianke, 2021-04-12:
OK, I accomplished line-circle, circle-circle tangent constrain, thank robnee :)
[image]
But a slightly more complicated shapes, for exmaple a square with round chamfer, can't use this method to create tangent constrain to other shapes, since the problem (4).
And since the problem (3), this method doesn't work with spline curves.
Thank you again :)

ruevs, 2021-04-12:
My guess is that what you are looking for is `Sketch | Tangent Ark at Point` or `Shift+A` .

ruevs, 2021-04-12:
I would suggest that you read through the reference (https://solvespace.com/ref.pl) or at least search for things that you need in it, e.g. "tangent".

The tutorials are very useful as well https://solvespace.com/tutorial.pl .

ruevs, 2021-04-12:
And your point 4 above is indeed valid. Albeit a rather corner case.

zhulaojianke, 2021-04-12:
Well, I'll keep learning and try. Thank you ruevs. 

The command "Tangent Ark at Point" (Shift + A)  has no effect on spline curves even at their endpoints.
[image]

I find a way to solve problem (3), although the constrained points cannot move independently along the curve.
[image]

zhulaojianke, 2021-04-21:
I still didn't give up trying to make a CAM, the constraint of the point on the curved surface. I have learned in the tutorial how to constrain a point to a cylinder surface, but it seems that what the only thing can I do about the curve surface is constrain a point to a cylinder surface. The constraint of a point on a surface is not only inapplicable to a cylinder surface extruded out by a closed contour composed of multiple line and curve segments, but is also invalid even on a simple sphere.

Here is my test document:
[point on curved surface constrain test.zip](https://github.com/solvespace/solvespace/files/6352867/point.on.curved.surface.constrain.test.zip)

Evil-Spirit, 2021-04-22:
There should be no more than 4 points in Bezier (two of them is for derivatives). And should be added two constraints:

a. 1st order Tangent 

b. 2nd order Tanget

I don't know where, but it was discussed here a lot.